aseba-community/aseba #540: Export as svg do not display the red dot in VPL

cbarraud, 2016-07-30:
Hi

I found a small bug in the export of VPL as svg. When you export your program, it does not display the small red dot beside ground or prox sensors. In bmp or png it's perfectly ok. 

Here is an example:

[image]

On the left, a small program exported in png, on the right, on svg.

When you open the svg (I opened it with Inkscape), you can see that there is some sort of small dot at the place where the red one should be if you ungroup the image a few times in a row:

[image]

It seems that the degraded of red is not export correctly.

I work on a Windows 7 PC.

Christophe

stephanemagnenat, 2016-08-02:
It is indeed a bug in the SVG exporter of Qt (hence foreign issue). It seems that it is not able to export the circular gradient. I suggest we document this bug on the wiki and close it, because there is nothing we can do for now, excepted writing a minimal test case and reporting it to Qt.

stephanemagnenat, 2016-12-21:
Maybe Qt 5 will solve this problem, so I tag this issue 1.7

ypiguet-epfl, 2018-02-23:
Or just draw a plain red dot without gradient when you export to SVG.