aseba-community/aseba #541: problème pda

bertrand432, 2016-08-03:
Bonjour,
j'essaie d'installer aseba sur ubuntu 3.13 mais la commande: sudo add-apt-repository ppa:stephane.magnenat/`lsb_release -c -s`
ne marche pas,
il ne reconnait pas le nom ou le format ppa
est-ce normal? y a t il un autre lien?
merci beaucoup
B Gonnet

stephanemagnenat, 2016-08-03:
Essayez de taper `lsb_release -c -s`, puis de copier/coller à la main le résultat après `sudo add-apt-repository ppa:stephane.magnenat/`, sans espace. Je pense qu'il y a une erreur de copier/coller de votre côté.

Lors de rapport de bogue, veuillez donner l'erreur exacte produite par le système, ainsi que les commandes qui ont conduit à cette erreur.

bertrand432, 2016-08-04:
Bonjour M Magnenat,
cela ne marche toujours pas,
je vous envoie une capture d'écran.
j'ai en fait ubuntu 14.04
merci de votre aide
B Gonnet

stephanemagnenat, 2016-08-04:
Votre capture d'écran n'a pas été attachée au rapport de bug. Normalement vous pouvez clisser-déposer un fichier dans la zone de texte d'un commentaire et il se fait attacher.

stephanemagnenat, 2016-12-05:
Ce problème est-il toujours présent ?

bertrand432, 2016-12-06:
Bonsoir,

en fait j'ai laissé tombé.

Je reprendrai les tests pendant les vacances de Noël.

Très cordialement

B Gonnet

bertrand432, 2017-01-01:
Bonsoir,
j'ai toujours le même souci.
Je vous joins une capture d'écran.
J'ai vu dans la doc qu'il faut brancher le robot avant de lancer le
logiciel.
Peut-on coder sans le robot?

Très cordialement et bonne année 2017

B Gonnet

bertrand432, 2017-01-02:
Bonjour,
j'ai pu installer le logiciel manuellement
cordialement
B Gonnet

closed 2017-01-04 by stephanemagnenat