aseba-community/aseba #544: asebadummynode should allow ephemeral ports (and advertise using Zeroconf)

davidjsherman, 2016-08-20:
Asebadummynode takes an optional argument 0–9 for its node type, and hardwires a TCP port in the range 33333–33342. This behavior is expected by many programs.

We should allow dynamic allocation of ephemeral ports as described in [RFC 6335 Internet Assigned Numbers Authority (IANA) Procedures for the Management of the Service Name and Transport Protocol Port Number Registry](https://tools.ietf.org/html/rfc6335#section-6). Since `bind` will do this in Posix and Windows if the requested port is 0, it should be enough to use `port=0` in the Dashel target.

I propose adding an optional parameter to asebadummynode:
* --port|-p INTEGER|any

to allow asebadummynode to request any port, with a value of 'any' or '0' indicating that a dynamic port number should be chosen from the ephemeral range. Note that this will also allow creating multiple instances of dummynode-_i_ with the same node id on the same host machine.

Since the caller will not know what port was allocated, asebadummynode could print the Dashel target, or at least the port number. Ideally, asebadummynode would advertise itself using Zeroconf as described in issue #543 “Advertise targets using Zeroconf”.

stephanemagnenat, 2016-08-22:
I agree. Also, the switch should have a way to list this port on HTTP. One way could be to list all Aseba targets plus the listening port. In JSON it is something like:
``{ asebaTargets = [ "ser:name=Thymio-II", "tcp:localhost;33335" ], asebaPort = 33333 }``

davidjsherman, 2016-08-23:
It looks like this is slightly tricky. A Dashel `tcpin:` stream will indeed receive a dynamically allocated ephemeral port if its target parameters contain `port=0`, but Dashel doesn't check what port was assigned. The way to find the port number is by calling `getsockname` (available in posix and winsock) on the file descriptor. This would require a modification to Dashel: we can't call `getsockname` ourselves, since the file descriptor is protected in Dashel::SelectableStream.

In Dashel, the constructors for SocketServerStream and UDPSocketStream could call `getsockname` after `bind` and record the assigned port in the parameter set.

A dynamic port number can't be reserved in advance, so we can't just reserve one and provide it to Dashel.

stephanemagnenat, 2016-08-24:
I've implemented the feature implemented in Dashel commit aseba-community/dashel@0686c134a48c858cc4aff86bb7c9c4827f0c9b75

davidjsherman, 2016-08-24:
I have a branch [zeroconf](https://github.com/davidjsherman/aseba/tree/zeroconf) with work in progress on an [Aseba::Zeroconf](https://github.com/davidjsherman/aseba/tree/zeroconf/common/zeroconf) object, starting with MacOS. [Asebadummynode](https://github.com/davidjsherman/aseba/blob/zeroconf/targets/dummy/dummynode.cpp) in 07900733ec5f7cb8e18a51091ba24cb14f3be8c5  announces itself, and can be reached by Aseba Studio under names like dummynode-0._aseba._tcp.local.

Here is an example starting three dummy nodes, the first two with the same name. By default the mDNS-SD service will invent a name when there is a conflict.
```
% asebadummynode --basePort 0 0
tcp:localhost;port=63382
% asebadummynode --basePort 0 0
tcp:localhost;port=63383
% asebadummynode --basePort 0 1
tcp:localhost;port=63384
```
Here are the DNS reports, showing the three PTR records for the zeroconf names, the three SRV records showing the port numbers, and the three TXT records showing the additional information. 
```
% dig @224.0.0.251 -p 5353 -t any _aseba._tcp.local

; <<>> DiG 9.8.3-P1 <<>> @224.0.0.251 -p 5353 -t any _aseba._tcp.local
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 18056
;; flags: qr aa; QUERY: 1, ANSWER: 3, AUTHORITY: 0, ADDITIONAL: 8

;; QUESTION SECTION:
;_aseba._tcp.local.		IN	ANY

;; ANSWER SECTION:
_aseba._tcp.local.	10	IN	PTR	dummynode-0._aseba._tcp.local.
_aseba._tcp.local.	10	IN	PTR	dummynode-0\032\(2\)._aseba._tcp.local.
_aseba._tcp.local.	10	IN	PTR	dummynode-1._aseba._tcp.local.

;; ADDITIONAL SECTION:
dummynode-0._aseba._tcp.local. 10 IN	SRV	0 0 63382 Oolong.local.
dummynode-0._aseba._tcp.local. 10 IN	TXT	"txtvers=1" "protovers=5" "type=Unknown" "ids=\000\001" "pids=\000\000"
dummynode-0\032\(2\)._aseba._tcp.local.	10 IN SRV 0 0 63383 Oolong.local.
dummynode-0\032\(2\)._aseba._tcp.local.	10 IN TXT "txtvers=1" "protovers=5" "type=Unknown" "ids=\000\001" "pids=\000\000"
dummynode-1._aseba._tcp.local. 10 IN	SRV	0 0 63384 Oolong.local.
dummynode-1._aseba._tcp.local. 10 IN	TXT	"txtvers=1" "protovers=5" "type=Unknown" "ids=\000\002" "pids=\000\000"
Oolong.local.		10	IN	AAAA	fe80::129a:ddff:feae:6c90
Oolong.local.		10	IN	A	10.0.0.102

;; Query time: 0 msec
;; SERVER: 10.0.0.102#5353(224.0.0.251)
;; WHEN: Wed Aug 24 20:28:53 2016
;; MSG SIZE  rcvd: 414
```

commit bb4643fb (2017-06-27): Added zeroconf support to dummynode, fix #544.
closed 2017-06-27 by stephanemagnenat