aseba-community/aseba #558: Aseba Studio and VPL: focus on only one node

davidjsherman, 2016-09-21:
When the connection to a robot is through a switch or through a shared RF dongle, other peer robot nodes on the network will appear in tabs in the Studio. In a classroom, the teacher may prefer that a student only see the robot assigned to her.

The proposal is to add a focus mode that **only shows a tab for a selected robot node**. One or all of the following might be useful:

1. A menu option for *focus mode* with a submenu of known robots
 - easiest, received node descriptions already in memory
1. Selecting one robot among several in the initial browser
 - depends on updated node browser
1.  "Beeping" the robot's QR code in front of the webcam
 - depends on e.g. [ZXing](https://github.com/zxing/zxing) (see [license question](https://github.com/zxing/zxing/wiki/License-Questions)) or [Zbar](http://zbar.sourceforge.net)
1. A command-line option, so that a specific desktop link could be created on each computer
 - easy, equivalent to Studio for Thymio, should fall back to browser if node not found

stephanemagnenat, 2016-09-21:
Aren't we solving the wrong problem? Shouldn't the robots be in different networks? If so, isn't the question to more easily reconfigure the networks?

But to be realistic, I guess we would rather focus our efforts on the Wi-Fi version, which solves this problem among others.

davidjsherman, 2016-09-21:
The question that came up is, what if the robots are sharing RF dongles.
Even with wifi, one can imagine an exercise where all of the robots in the classroom are in the same network, because they must communicate or because they must all see broadcast messages from the teacher (think choreography, different dancers with the same music). The teacher might not want to allow the students to modify their neighbors' robots.

stephanemagnenat, 2016-09-21:
> Even with wifi, one can imagine an exercise where all of the robots in the classroom are in the same network, because they must communicate or because they must all see broadcast messages from the [...]

Everyone keeps talking about this idea of having multiple robots in a network, but beside the simple remote-control demo using two Wireless Thymios, I have yet to see a concrete application. Moreover, I doubt of the scalability of the system to having a dozen of robots in the same network.

I think that it is better to focus on a simple use case in which every user program one robot, and each robot is in its own network.

Regarding usage in classrooms, I think that there is an important question of allowing the teacher to control what is going on, but this question is larger than the one of the access of Studio to a robot. For instance, the teacher might want to stop all robots for a while so that children listen to her/him.

stephanemagnenat, 2016-09-21:
That said, implementing the selection of a specific robot for VPL standalone using the combined information from zeroconf and scanning the network is planned as discussed, which corresponds to option 2. But I do not want to add complex access control to Studio.