aseba-community/aseba #560: Branch http-refactor does not compile

stephanemagnenat, 2016-09-28:
After a merge with master, the branch `http-refactor` does not compile any more, there is an error in `targetadvertise.cpp`:
```
In file included from /home/steph/data/aseba-thymio/aseba-community/aseba/examples/zeroconf/targetadvertise.cpp:22:
In file included from /home/steph/data/epfl-lsro/aseba-community/dashel/dashel/dashel.h:44:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/string:41:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/allocator.h:46:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/x86_64-linux-gnu/c++/5.4.0/bits/c++allocator.h:33:
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/ext/new_allocator.h:120:23: error: call to implicitly-deleted copy constructor of 'std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node,
      std::default_delete<Aseba::NodesManager::Node> > >'
        { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
                             ^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/alloc_traits.h:530:8: note: in instantiation of function template specialization '__gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > > >::construct<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> >
      >, const std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > &>' requested here
        { __a.construct(__p, std::forward<_Args>(__args)...); }
              ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:529:23: note: in instantiation of function template specialization 'std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<const unsigned
      int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > > > >::construct<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node,
      std::default_delete<Aseba::NodesManager::Node> > >, const std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > &>' requested here
              _Alloc_traits::construct(_M_get_Node_allocator(),
                             ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:546:4: note: in instantiation of function template specialization 'std::_Rb_tree<unsigned int, std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > >, std::_Select1st<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node>
      > > >, std::less<unsigned int>, std::allocator<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > > >::_M_construct_node<const std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > &>' requested here
          _M_construct_node(__tmp, std::forward<_Args>(__args)...);
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:459:18: note: in instantiation of function template specialization 'std::_Rb_tree<unsigned int, std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > >, std::_Select1st<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node>
      > > >, std::less<unsigned int>, std::allocator<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > > >::_M_create_node<const std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > &>' requested here
          { return _M_t._M_create_node(_GLIBCXX_FORWARD(_Arg, __arg)); }
                        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:569:23: note: in instantiation of function template specialization 'std::_Rb_tree<unsigned int, std::pair<const unsigned int,
      std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > >, std::_Select1st<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node>
      > > >, std::less<unsigned int>, std::allocator<std::pair<const unsigned int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > > >::_Alloc_node::operator()<const std::pair<const unsigned
      int, std::unique_ptr<Aseba::NodesManager::Node, std::default_delete<Aseba::NodesManager::Node> > > &>' requested here
          _Link_type __tmp = __node_gen(*__x->_M_valptr());
                             ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:1574:21: note: (skipping 6 contexts in backtrace; use -ftemplate-backtrace-limit=0 to see all)
        _Link_type __top = _M_clone_node(__x, __node_gen);
                           ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:529:23: note: in instantiation of function template specialization 'std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<Dashel::Stream
      *const, Aseba::StreamNodesManager> > > >::construct<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager>, Dashel::Stream *&, Aseba::StreamNodesManager>' requested here
              _Alloc_traits::construct(_M_get_Node_allocator(),
                             ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:546:4: note: in instantiation of function template specialization 'std::_Rb_tree<Dashel::Stream *, std::pair<Dashel::Stream *const,
      Aseba::StreamNodesManager>, std::_Select1st<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >, std::less<Dashel::Stream *>, std::allocator<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >
      >::_M_construct_node<Dashel::Stream *&, Aseba::StreamNodesManager>' requested here
          _M_construct_node(__tmp, std::forward<_Args>(__args)...);
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_tree.h:2123:19: note: in instantiation of function template specialization 'std::_Rb_tree<Dashel::Stream *, std::pair<Dashel::Stream *const,
      Aseba::StreamNodesManager>, std::_Select1st<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >, std::less<Dashel::Stream *>, std::allocator<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >
      >::_M_create_node<Dashel::Stream *&, Aseba::StreamNodesManager>' requested here
        _Link_type __z = _M_create_node(std::forward<_Args>(__args)...);
                         ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_map.h:559:16: note: in instantiation of function template specialization 'std::_Rb_tree<Dashel::Stream *, std::pair<Dashel::Stream *const,
      Aseba::StreamNodesManager>, std::_Select1st<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >, std::less<Dashel::Stream *>, std::allocator<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> >
      >::_M_emplace_unique<Dashel::Stream *&, Aseba::StreamNodesManager>' requested here
        { return _M_t._M_emplace_unique(std::forward<_Args>(__args)...); }
                      ^
/home/steph/data/aseba-thymio/aseba-community/aseba/examples/zeroconf/targetadvertise.cpp:139:17: note: in instantiation of function template specialization 'std::map<Dashel::Stream *, Aseba::StreamNodesManager, std::less<Dashel::Stream
      *>, std::allocator<std::pair<Dashel::Stream *const, Aseba::StreamNodesManager> > >::emplace<Dashel::Stream *&, Aseba::StreamNodesManager>' requested here
                                                streamMap.emplace(cxn,StreamNodesManager(cxn));
                                                          ^
[... cut ...]
      _T2 second;                /// @c second is a copy of the second object
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:356:7: note: 'unique_ptr' has been explicitly marked deleted here
      unique_ptr(const unique_ptr&) = delete;
      ^
```

It seems that the `emplace()` at line 139 does not work because it is forbidden to copy-construct a pair of unsigned to `unique_ptr<Aseba::NodesManager::Node>`. I do not understand what triggers this error. @davidjsherman do you have an idea?

stephanemagnenat, 2016-09-28:
Commit 3ee5fbf, which uses a `unique_ptr<StreamNodesManager>` rather than a `StreamNodesManager` does correct the compilation problem, but why would it work before?

closed 2016-09-28 by stephanemagnenat

davidjsherman, 2016-09-28:
Neither GNU 6.1.1 (Debian), GNU 4.9.4 (Ubuntu), GNU 5.3.0 (Win7) nor AppleClang 6.0.0.6000051 (OS X) saw this as an error when compiling together
* Dashel davidjsherman/dashel@474435c132694c5ca4d546d8e28fd79220e83106 (pollsocketstream) if getOrCreateSocket created the socket, ~PollStream will have to clo…
* Enki enki-community/enki@8bc1f2abdf1ec919e1e9d9c363eb2a5e300de3b4 (master) Added callback for collision
* Aseba davidjsherman/aseba@f379417dda8371abc7ee42d8a134d5a136f6afe1 (zeroconf-PR-derived) don't use for loop initial declarations

I don't see a pertinent difference in https://github.com/aseba-community/aseba/compare/3ee5fbfc787c7aef2ac5d27a03c153a9eb999351...davidjsherman:f379417dda8371abc7ee42d8a134d5a136f6afe1.

stephanemagnenat, 2016-09-28:
I do not understand your comment. Here on Ubuntu 16.04.1 (GCC 5.4.0 and clang 3.8.0-2ubuntu4) commit 3ee5fbfc787c7aef2ac5d27a03c153a9eb999351 was necessary for the http-refactor branch to compile.