aseba-community/aseba #563: Aseba Studio reseting connection to thymio

aobrodsky, 2016-10-18:
Sumary: Aseba Studio  1.5.3 seems to reset connection to Thymio (both over USB and network)

To reproduce:
  Launch Aseba Studio and connect to Thymio
  Create simple program (say move forward)
  Click on "auto" checkbox
  Click on "Load"
  Click on "Run", which becomes "Pause"

  At some point later Aseba Studion UI blinks.  The "auto" is unset.  The "Pause" resets to "Run" and
  the button is disabled. I.e., as if Studio just reconnected to the robot. 

Environment: Mac OS X 10.12 Sierra on 27" iMac [Late 2013], 32 GB of ram.

ttyl
Alex

stephanemagnenat, 2016-10-21:
Hello,

Are you using the USB cable or a wireless Thymio?

kind regards

aobrodsky, 2016-10-25:
Hi,

This was a USB Thymio.  However, I plugged in another one yesterday, and the behavior has ceased.  I’ll play around with the USB thymios and the network based connection and see if I can make the problem more reproducible,

ttyl
Alex

> 
> Hello,
> 
> Are you using the USB cable or a wireless Thymio?
> 
> kind regards

------------------------------------------------------------------------------
To put it bluntly, we simply do not know yet what we should be talking about,
but that should not worry us, for it just illustrates what was meant by
"intangible goods and uncertain rewards".
                           -- Edsger Dijkstra, 'The End of Computer Science'

stephanemagnenat, 2016-10-26:
It might be that there is a soldering problem on the USB connector of your first robot. In that case, you want to return that robot for repairs. Did you buy it at a shop or directly from Mobsya?

aobrodsky, 2016-10-26:
Hi,

It also happens when I am connected to a Thymio simulator (over the network).  It may be a time-out issue of some kind, but I have not looked into the dashel code to investigate.

ttyl
Alex

> 
> It might be that there is a soldering problem on the USB connector of your first robot. In that case, you want to return that robot for repairs. Did you buy it at a shop or directly from Mobsya?

------------------------------------------------------------------------------
To put it bluntly, we simply do not know yet what we should be talking about,
but that should not worry us, for it just illustrates what was meant by
"intangible goods and uncertain rewards".
                           -- Edsger Dijkstra, 'The End of Computer Science'

aobrodsky, 2016-10-27:
Hi,

Investigating this problem further, it seems the issue only has recently arose under the new Mac OS X Sierra.

The issue occurs when the Aseba Studio (AS) is connected to either a Thymio or simulators (via a network connection).   The amount of time that it takes for the issue to manifest itself varies, but it does seem to occur inevitably.  I.e., in some cases it can occur within minutes, in other cases within hours.

The connection itself is not reset (or dropped).  What seems to happen is that the Aseba Studio UI resets in the same manner as when a disconnection / reconnection occurs.  But, at the lower level no disconnection actually occurs.   I.e., When AS is connected to a simulator over a network, the network connection itself is never closed, and AS continues to use the same network connection after it resets.  

As I am writing this email, I have AS running in the background and it seems to be reseting itself every couple minutes.

This tends to happen when AS is not the foreground task.  I.e., if one of the other apps has the focus).  I have not dug into AS / dashel code to see where this is happening.

ttyl
Alex

> 
> Hi,
> 
> It also happens when I am connected to a Thymio simulator (over the network).  It may be a time-out issue of some kind, but I have not looked into the dashel code to investigate.
> 
> ttyl
> Alex
> 
>> 
>> It might be that there is a soldering problem on the USB connector of your first robot. In that case, you want to return that robot for repairs. Did you buy it at a shop or directly from Mobsya?
>> 
>> 
> 
> ------------------------------------------------------------------------------
> To put it bluntly, we simply do not know yet what we should be talking about,
> but that should not worry us, for it just illustrates what was meant by
> "intangible goods and uncertain rewards".
>                           -- Edsger Dijkstra, 'The End of Computer Science'

------------------------------------------------------------------------------
To put it bluntly, we simply do not know yet what we should be talking about,
but that should not worry us, for it just illustrates what was meant by
"intangible goods and uncertain rewards".
                           -- Edsger Dijkstra, 'The End of Computer Science'

aobrodsky, 2016-10-27:
Hi,

Here is what I suspect may be happening.  

NodeManager in Aseba Studio disconnects from Nodes that it has not heard form in more than 3 seconds.  It may be that when AS is in the background, it may be suspended for more than 3 seconds, resulting in a disconnect followed by a reconnect of target.  This will result through a nodeDisconnect / modeConnect signals a teardown and instantiation of the NodeTab for the target, without the underlying connection actually being dropped /reset.

I have not confirmed this, since I have not recompiled AS from source.

ttyl
Alex

> 
> Hi,
> 
> Investigating this problem further, it seems the issue only has recently arose under the new Mac OS X Sierra.
> 
> The issue occurs when the Aseba Studio (AS) is connected to either a Thymio or simulators (via a network connection).   The amount of time that it takes for the issue to manifest itself varies, but it does seem to occur inevitably.  I.e., in some cases it can occur within minutes, in other cases within hours.
> 
> The connection itself is not reset (or dropped).  What seems to happen is that the Aseba Studio UI resets in the same manner as when a disconnection / reconnection occurs.  But, at the lower level no disconnection actually occurs.   I.e., When AS is connected to a simulator over a network, the network connection itself is never closed, and AS continues to use the same network connection after it resets.  
> 
> As I am writing this email, I have AS running in the background and it seems to be reseting itself every couple minutes.
> 
> This tends to happen when AS is not the foreground task.  I.e., if one of the other apps has the focus).  I have not dug into AS / dashel code to see where this is happening.
> 
> ttyl
> Alex
> 
> 
>> 
>> Hi,
>> 
>> It also happens when I am connected to a Thymio simulator (over the network).  It may be a time-out issue of some kind, but I have not looked into the dashel code to investigate.
>> 
>> ttyl
>> Alex
>> 
>>> 
>>> It might be that there is a soldering problem on the USB connector of your first robot. In that case, you want to return that robot for repairs. Did you buy it at a shop or directly from Mobsya?
>>> 
>>> 
>> 
>> ------------------------------------------------------------------------------
>> To put it bluntly, we simply do not know yet what we should be talking about,
>> but that should not worry us, for it just illustrates what was meant by
>> "intangible goods and uncertain rewards".
>>                          -- Edsger Dijkstra, 'The End of Computer Science'
>> 
> 
> ------------------------------------------------------------------------------
> To put it bluntly, we simply do not know yet what we should be talking about,
> but that should not worry us, for it just illustrates what was meant by
> "intangible goods and uncertain rewards".
>                           -- Edsger Dijkstra, 'The End of Computer Science'

------------------------------------------------------------------------------
To put it bluntly, we simply do not know yet what we should be talking about,
but that should not worry us, for it just illustrates what was meant by
"intangible goods and uncertain rewards".
                           -- Edsger Dijkstra, 'The End of Computer Science'

stephanemagnenat, 2016-10-28:
I think you are totally right. I guess that there is a new power-saving feature in OS X Sierra that suspends background applications. Maybe it does so only for targets having an OpenGL context? Could you try with asebadummynode, to see if the same problem happens? Thank you!

aobrodsky, 2016-10-28:
Hi,

Confirmed.  Happens with asebadummynode as well.  I think it;s the clients (asebastudio) rather than the targets (thymio, asebadumynode, my thymio simulator).  This could be a problem within Qt rather than asebastudio per se, 

In the long term, one solution may be to query nodes that have not been heard from and then disconnect after a second time-out, kind of a second chance approach.  In the short term, increasing the time-out to 6 seconds may reduce the frequency of these resets, but tat’s a bit of a hack. :)

ttyl
Alex

> 
> I think you are totally right. I guess that there is a new power-saving feature in OS X Sierra that suspends background applications. Maybe it does so only for targets having an OpenGL context? Could you try with asebadummynode, to see if the same problem happens? Thank you!

------------------------------------------------------------------------------
To put it bluntly, we simply do not know yet what we should be talking about,
but that should not worry us, for it just illustrates what was meant by
"intangible goods and uncertain rewards".
                           -- Edsger Dijkstra, 'The End of Computer Science'

aobrodsky, 2016-10-28:
Hi,

Easy fix to this problem.  The installer should turn App Nap off for Aseba Studio.  For current installs the following command-line in Terminal will work:

  defaults write org.mobsya.asebastudio NSAppSleepDisabled -bool YES

The corresponding preferences plist for AS should look something like this:

?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>NSAppSleepDisabled</key>
        <true/>
        <key>NSNavPanelExpandedSizeForOpenMode</key>
        <string>{712, 459}</string>
        <key>NSNavPanelExpandedSizeForSaveMode</key>
        <string>{712, 532}</string>
        <key>NSNavPanelExpandedStateForSaveMode</key>
        <true/>
</dict>
</plist>

Note the first entry in the top-level dictionary.  The fix seems to work, but I have not tested it extensively.

ttyl
Alex

stephanemagnenat, 2016-11-16:
Fixed in aseba-community/aseba-osx@254f81e5a0a52c07b1e7e7ddce8b82cc470d3148.

Please test and close.

aobrodsky, 2016-11-17:
Tested. Fix works.

closed 2016-11-17 by aobrodsky