aseba-community/aseba #568: Resurct build server for Windows and OS X builds on release-1.5.x branch

stephanemagnenat, 2016-11-02:
Necessary to release 1.5.4

davidjsherman, 2016-11-15:
Currently:

1. Building and packaging on ci.inria.fr does work for macOS and Windows
1. Branch [1-buildfarm](https://github.com/davidjsherman/aseba-jenkins/tree/1-buildfarm) in [davidjsherman/aseba-jenkins](https://github.com/davidjsherman/aseba-jenkins) has Ansible recipes for creating build environments for Ubuntu, macOS, and Windows
1. I have most of a packager script for Debian, using *pbuilder* with a hook to a bind-mounted local repo to avoid global installation of dashes and enki on the build host

To do:

1. Replace ad hoc packager scripts with cpack
1. Use hard links in macOS app bundles to that they can be copied and moved around independently
1. Write a master Jenkinsfile to automate build, test, package pipelines
1. Configure Jenkins to also automatically compile pull requests

stephanemagnenat, 2016-11-16:
Done.

closed 2016-11-16 by stephanemagnenat

stephanemagnenat, 2016-12-15:
Created issue #592 to track progress of the move of CI to Inria.