aseba-community/aseba #573: Blockly crash (from the thymio forum)

cbarraud, 2016-11-22:
Hi,

I just transmit a message from Thymio's forum: https://www.thymio.org/forum/t-1982307/blockly-plantage-lors-de-l-utilisation-de-repeter-jusqu-a#post-2685986

-------------------------------

Bonjour
Ne trouvant pas de forum dédié à blockly, je poste dans celui concernant VPL, pour un crash de l'interface lors de l'utilisation de blockly.
Ce comportement a été constaté avec la version 1.5.3 d'AsebaStudio sur 2 ordinateurs différents (un sous Debian testing, l'autre sous Ubuntu).
Pour reproduire le bug, il suffit d'essayer de construire une boucle du type
"répéter jusqu'à (capteur de sol détecte quelque chose et vrai)" :
[image]

L'interface plante alors avec un SIGABRT. Voici la stacktrace récupérée par le test du bug sous gdb:

Starting program: /usr/bin/asebastudio 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8336700 (LWP 7267)]
[New Thread 0x7fffe7b35700 (LWP 7268)]
[Thread 0x7fffe8336700 (LWP 7267) exited]
[Thread 0x7fffe7b35700 (LWP 7268) exited]
[New Thread 0x7fffe7b35700 (LWP 7271)]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fffe8336700 (LWP 7272)]
[New Thread 0x7fffdffff700 (LWP 7273)]
[Thread 0x7fffe8336700 (LWP 7272) exited]
[Thread 0x7fffdffff700 (LWP 7273) exited]
[New Thread 0x7fffdffff700 (LWP 7274)]
[New Thread 0x7fffe8336700 (LWP 7275)]
[New Thread 0x7fffde7eb700 (LWP 7276)]
[New Thread 0x7fff9787b700 (LWP 7281)]

Thread 10 "Thread (pooled)" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff9787b700 (LWP 7281)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
58    ../sysdeps/unix/sysv/linux/raise.c: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x00007ffff314940a in __GI_abort () at abort.c:89
#2  0x000000000053f767 in Aseba::BinaryArithmeticNode::deMorganNotRemoval() ()
#3  0x000000000053fa97 in Aseba::UnaryArithmeticNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#4  0x000000000053fb30 in Aseba::WhileNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#5  0x000000000053feb4 in Aseba::BlockNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#6  0x000000000053ff67 in Aseba::IfWhenNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#7  0x000000000053feb4 in Aseba::BlockNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#8  0x0000000000523c73 in Aseba::Compiler::compile(std::basic_istream<wchar_t, std::char_traits<wchar_t> >&, Aseba::BytecodeVector&, unsigned int&, Aseba::Error&, std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#9  0x000000000049c157 in Aseba::compilationThread (targetDescription=..., commonDefinitions=..., 
    source=..., dump=dump@entry=true)
    at /build/aseba-EiRPC2/aseba-1.5.3/clients/studio/MainWindow.cpp:1059
#10 0x00000000004b8d74 in QtConcurrent::StoredFunctorCall4<Aseba::NodeTab::CompilationResult*, Aseba::NodeTab::CompilationResult* (*)(Aseba::TargetDescription, Aseba::CommonDefinitions, QString, bool), Aseba::TargetDescription, Aseba::CommonDefinitions, QString, bool>::runFunctor (this=0xf89b90)
    at /usr/include/qt4/QtCore/qtconcurrentstoredfunctioncall.h:876
#11 0x00000000004b3cc5 in QtConcurrent::RunFunctionTask<Aseba::NodeTab::CompilationResult*>::run (
    this=0xf89b90) at /usr/include/qt4/QtCore/qtconcurrentrunbase.h:106
#12 0x00007ffff3dbcdba in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#13 0x00007ffff3dc9daa in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#14 0x00007ffff2cf7464 in start_thread (arg=0x7fff9787b700) at pthread_create.c:333
#15 0x00007ffff31fd9df in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:105
(gdb)

-------------------------------

On my platform (Windows 8.1, Aseba 1.5.3) it crashes also:

[image]

It's really when you place the sensor (whichever) that it crashes.

I'll put a reference to this issue in the forum.

Christophe

stephanemagnenat, 2016-11-24:
Can you please try with version 1.5.4? In general, it is good to try with the latest stable pre-release to avoid opening bugs that were fixed recently.

stephanemagnenat, 2016-11-24:
I cannot reproduce with master, it is most likely linked to a corrected compiler bug, please make sure with 1.5.4 that it is fixed and close the bug, thank you!

oaubert, 2016-11-27:
Upgrading to 1.5.4 does not fix the issue, I have the exact same behaviour: it does not crash if I use a "repeat while" loop, it does not crash if I use "repeat until sensor * detects *" but it crashes if I use "repeat until sensor * detects * and true". I crashes in multiple construction cases:
- put a "repeat until", add a "and", then put a "sensor * detects" -> crash
- put a "repeat while", add a "and", then put a "sensor * detects" (still OK there), switch the "while" to "until" -> crash

Here is the backtrace:
[code]
Thread 10 "Thread (pooled)" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff96fd5700 (LWP 7498)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
58	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x00007ffff314940a in __GI_abort () at abort.c:89
#2  0x00005555556b48a7 in Aseba::BinaryArithmeticNode::deMorganNotRemoval() ()
#3  0x00005555556b4bea in Aseba::UnaryArithmeticNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#4  0x00005555556b5280 in Aseba::WhileNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#5  0x00005555556b4c5a in Aseba::BlockNode::optimize(std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#6  0x00005555556996ad in Aseba::Compiler::compile(std::basic_istream<wchar_t, std::char_traits<wchar_t> >&, Aseba::BytecodeVector&, unsigned int&, Aseba::Error&, std::basic_ostream<wchar_t, std::char_traits<wchar_t> >*) ()
#7  0x00005555556115b1 in Aseba::compilationThread (targetDescription=..., commonDefinitions=..., source=..., dump=dump@entry=true)
    at /build/aseba-iswC_F/aseba-1.5.4/clients/studio/MainWindow.cpp:1059
#8  0x000055555562d794 in QtConcurrent::StoredFunctorCall4<Aseba::NodeTab::CompilationResult*, Aseba::NodeTab::CompilationResult* (*)(Aseba::TargetDescription, Aseba::CommonDefinitions, QString, bool), Aseba::TargetDescription, Aseba::CommonDefinitions, QString, bool>::runFunctor (this=0x555556a2edc0)
    at /usr/include/qt4/QtCore/qtconcurrentstoredfunctioncall.h:876
#9  0x0000555555629e95 in QtConcurrent::RunFunctionTask<Aseba::NodeTab::CompilationResult*>::run (this=0x555556a2edc0)
    at /usr/include/qt4/QtCore/qtconcurrentrunbase.h:106
#10 0x00007ffff3dbcdba in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#11 0x00007ffff3dc9daa in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#12 0x00007ffff2cf7464 in start_thread (arg=0x7fff96fd5700) at pthread_create.c:333
#13 0x00007ffff31fd9df in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:105
(gdb) 
[/code]

stephanemagnenat, 2016-11-28:
We will try to reproduce this bug and fix it before 1.5.4 release.

stephanemagnenat, 2016-11-28:
Bug confirmed on Linux in 1.5.4

stephanemagnenat, 2016-11-28:
I have back-ported the correct optimisation code from master to release-1.5.4 and added a unit test. Therefore, this bug should be fixed.

closed 2016-11-28 by stephanemagnenat