aseba-community/aseba #575: strange behavior sometimes with blockly

mbonani, 2016-11-23:
I just see in live demo of Francesco the temperature sensor block that was frozen. Even if we close the blocky windows,  when we reopen blocky, the block was still there and froze the compiler.
Restart studio resolve the problem
No receipt to reproduce the bug... sorry.

stephanemagnenat, 2016-11-23:
Which version?

mbonani, 2016-11-23:
1.53 release. Perhaps fixed.

stephanemagnenat, 2016-11-23:
Please test with 1.5.4

mbonani, 2016-11-23:
cannot reproduce with windows 1.5.4. But bug was on a mac OSX with 1.5.3. I let you decide if we can close.

stephanemagnenat, 2016-11-24:
Let's keep it open until we have the final 1.5.4 package for OS X and let's test with that one.

stephanemagnenat, 2016-12-05:
Could you try to reproduce this problem with latest OS X packages?

stephanemagnenat, 2017-07-07:
@mbonani can you check whether this issue is still open.

cor3ntin, 2017-12-07:
Blockly is being removed from studio.

closed 2017-12-07 by cor3ntin