aseba-community/aseba #576: blockly put motor speed using variable and run backwards

mbonani, 2016-11-23:
There is a bug when you use a variable instead of a value when you use the block "run motor **backward** with speed _variable"_

[image]

it result with this code:

```
onevent button.backward
	when button.backward == 1 do
		v = v - 30
		motor.left.target = NaN
		motor.right.target = NaN
	end

```

It work with numerical value. 
Test on 1.5.4

stephanemagnenat, 2016-11-24:
That's definitely a bug, could you please share/send the corresponding aesl? Thank you!

stephanemagnenat, 2016-11-24:
Bug narrowed down to `generators/aesl/thymio.js`: 373 in blockly, the problem is that the value from the subsequent block is negated, but actually it should be composed with the '-' string as the goal is to generate code.

I'm working on a correction for the release branch.

stephanemagnenat, 2016-11-24:
Fixed in commit aseba-community/blockly@d6f4cfd546956761754847e99044adfbb907f9e6

closed 2016-11-24 by stephanemagnenat