aseba-community/aseba #577: Display bug for ELSE in the IF condition

FrancescoMondada, 2016-11-24:
When taking the IF block in blockly, activating the ELSE condition is very complex and with display bug on MACOSX (10.8.5) with aseba studio 1.5.3
[image]

mbonani, 2016-12-02:
I confirm on different OSX I have tested this strange black display bug. Nothing on Windows

stephanemagnenat, 2016-12-05:
This needs to be tested with the new Qt WebEngine which is available in recent Qt 5 distributions.

cor3ntin, 2017-12-07:
The development of blockly is now happening in

https://github.com/aseba-community/blockly
https://github.com/aseba-community/thymio-blockly-standalone

closed 2017-12-07 by cor3ntin