aseba-community/aseba #578: Srolling the menu window generates a selection

FrancescoMondada, 2016-11-24:
On MACOSX (10.8.5) and Blocly included in Studio 1.5.3, there is a problem of selection of the items in the menus of commands. To be reproduced: start studio, activate blockly, click on one of the menus of orders (for instance "actioneurs", point on one of the item (the item is activated) and scroll (using the trackpad). The item is selected and placed.
The scroll functionality has also strage behaviors on the main window...

mbonani, 2016-12-02:
Do not happen in Windows, 
But the scrolling is not working at all. No reaction in blockly when you scroll.

stephanemagnenat, 2016-12-05:
This needs to be tested with the new Qt WebEngine which is available in recent Qt 5 distributions.

stephanemagnenat, 2017-07-07:
It would be interesting to run the current blockly code in a Qt WebEngine widget in Qt5 to see if this bug has disappeared.

cor3ntin, 2017-12-07:
The development of blockly is now happening in

https://github.com/aseba-community/blockly
https://github.com/aseba-community/thymio-blockly-standalone

closed 2017-12-07 by cor3ntin