aseba-community/aseba #583: Native function for fast vector Deque

davidjsherman, 2016-12-01:
Sequential programming with the Aseba language is fastidious. Because the VM doesn't support blocking operations, preemptive scheduling, or a call stack, the approach in [`thymio_motion.aesl `](https://github.com/davidjsherman/inirobot-scratch-thymioII/blob/master/thymio_motion.aesl) is to use a queue that is polled at 100 Hz by a `motor` event. The queue is implemented using arrays and index variables, with insertion handled by scanning forward through the arrays. This *ad hoc* solution is complicated, fragile, and inefficient.

Native functions to implement a fast vector deque would make it easier to implement sequential operations in Aseba programs, and reduce bytecode size. The implementation could be based on [ArrayDeque: Fast Deque Operations Using an Array](http://opendatastructures.org/ods-java/2_4_ArrayDeque_Fast_Deque_O.html). Four operations would be defined:

* `deque.get(A, B, i)` Copy *len*`B` elements from position `i * `*len*`B` in `A`
* `deque.set(A, B, i)` Assign *len*`B` elements to position `i * `*len*`B` in `A`
* `deque.add(A, B, i)` Insert *len*`B` elements at position `i * `*len*`B` in `A`, shifting the existing contents to make room (Figure 2.3). Elements are lost from the right if `A` is full.
* `deque.remove(A, i, n)` Remove `n` elements at position `i * n` in `A`, shifting the existing contents (Figure 2.3).

(The first two elements of `A` are reserved for the `j` and `n` pointers, which is why `get` and `set` aren't simply implemented using existing Aseba vector addressing operations.)

Implementing a simple motion queue would give something like:
```
var Queue[122] # Store up to 40 operations e.g. for BeeBot
var operation[3]

sub motion_add
call deque.add(Queue,operation[0:2],-1) # push at back

onevent timer0
call deque.get(Queue,operation,0) 
timer.period[0] = operation[0]
motor.left.target = operation[1]
motor.right.target = operation[2]
deque.remove(Queue,0,3) # pop from front
```

stephanemagnenat, 2016-12-01:
I like it! @mbonani do we still have some flash space left in the Thymio firmware to add some native functions?

stephanemagnenat, 2016-12-01:
About the API, I understand that negative positions are counting from the back, which is not supported in the referenced document, right?

Do you also plan to have the operations `push_back`, `push_front`, `pop_back`, and `pop_front`, for ease of use?

stephanemagnenat, 2016-12-01:
> (The first two elements of A are reserved for the j and n pointers, which is why get and set aren't simply implemented using existing Aseba vector addressing operations.)

Isn't the reason because the data are shifting within the array? Because otherwise we could put `j` and `n` at the end of the array and still use Aseba opertations, just we would not detect the overwrite of these indices.

davidjsherman, 2016-12-01:
At the end is better because zero-based index arithmetic would work. It would allow unaligned reading (for example reading a sliding window whose size is different from *len*`B`). However using Aseba operations still requires knowing in the `j` offset
The reference doesn't have negative positions because it assumes that one can just subtract from `a.n`, typically `push_back` is just `a.add(a.end(),x)`, but we can't do nested calls in Aseba. I guess convenience functions are cleaner. That would give:
* `deque.get(A, B, i)` equivalent to `B = A[((j+i)*`*len*`B)%(A[122]-A[121]):(`*len*`B-1)*((j+i)*`*len*`B)%(A[122]-A[121])]`?
* `deque.set(A, B, i)` as above but `A[`...`] = B`
* `deque.add(A, B, i)`
* `deque.remove(A, i, n)`
* `deque.push_back(A, B)` syntactic sugar for `deque.add(A, B, -1)`
* `deque.push_front(A, B)` syntactic sugar for `deque.add(A, B, 0)`
* `deque.pop_back(A, n)` syntactic sugar for `deque.remove(A, -n, n)`
* `deque.pop_front(A, n)` syntactic sugar for `deque.remove(A, 0, n)`

`deque.add` and `deque.remove` are the only truly nontrivial operations. It might add 8% to the firmware size.

davidjsherman, 2016-12-02:
If we want to allow users to use Aseba operations on deques, it might be more convenient for them to know where the `j` offset is, and to store the index of the first element rather than the number of tuples. Even if they have to multiply by the tuple length.

if `j` is stored at `A[0]`, then
* `deque.get(A, B, i)` ≈ `B = A[A[0]+i*LEN:A[0]+(i+1)*LEN-1]`

and, more usefully,
* `deque.get(A, B, 0)` ≈ `B = A[A[0]:A[0]+LEN-1]`

where `LEN` is a constant for the length of tuples `B`.

This is still complicated enough that having `get` and `set` convenience operations is good, if there is room in the firmware.

Since the length of `A` is fixed and a read out of bounds should raise an exception.

mbonani, 2016-12-02:
D:/asebacomp/aseba-target-thymio2/Thymio2Firmware.X
Thymio2Firmware
Project Type: Application - Configuration: default
Device
PIC24FJ128GB106
 Checksum: Blank, no code loaded
Compiler Toolchain
XC16 (v1.26) [C:\Program Files (x86)\Microchip\xc16\v1.26\bin]
Production Image: Optimization: gcc s
Memory
Usage Symbols disabled. Click to enable Load Symbols.
Data 16384 (0x4000) bytes
Data Used: 69.4%
Data Used: 11364 (0x2C64) Free: 5020 (0x139C)
Program 83460 (0x14604) words
Program Used: 79.9%
Program Used: 66720 (0x104A0) Free: 16740 (0x4164)

I am not sure the settings space is inculded...

Michael Bonani

Association Mobsya
Ch. du Closel 5
1020 Renens

Tél: +41 (0)32 511 64 93

stephanemagnenat, 2016-12-04:
> `deque.pop_back(A, n)` syntactic sugar for `deque.remove(A, -n, n)`

Shouldn't `pop_back` copy the popped elements into a variable, to create a symmetry to `push_back`? Also because otherwise we would need an accessor called `deque.back(A, B)`. If `pop_back` copies the popped data an example call would be `deque.pop_back(A, B)` and retrieve the number of elements to pop from `length(B)`.

> Would it be clearer for young programmers if we used the C++ names for the operations, `insert` instead of `add` and `erase` instead of `remove`?

Yes I think it would be better, and also a slightly bit more precise imho.

> If we want to allow users to use Aseba operations on deques, it might be more convenient for them to know where the `j` offset is, and to store the index of the first element rather than the number of tuples. Even if they have to multiply by the tuple length.

I prefer that the multiplication is done inside the function as in your original proposal. With that, the code reads like we would have a proper container library which I think will help to make readable codes and find bugs. There are still some open questions (assuming `A` and `B` having the same semantics as in your example):
- Do we through an exception in any call if `length(A) % length(B) != 0`?
- When going `get`/`set` out of bounds, do we through the "normal" Aseba exception, or a specific one to this function. If we through the normal one, which index do we use in it (the raw or the multiplied one)?
- What do we do when `insert`ing into a full queue? And out of queue?
- What do we do when `erasing`ing an empty queue? And out of queue?
- What do we do when `push_back/front`ing into a full queue?
- What do we do when `pop_back/front`ing from an empty queue?

davidjsherman, 2016-12-04:
To have something more complete to discuss, I made a preliminary implementation in [PR 585](https://github.com/aseba-community/aseba/pull/585). It was developed starting from 25 tests I added to `tests/vm/data`.

It doesn't do size multiplication, because I was unsure whether it was always possible to infer the tuple size. For `get`, `set`, `insert`, and `push` it can be inferred from the size of template parameter n° 2. For `erase` and `pop` it must be a parameter. We could store it in the array but I think memory is scarce, isn't it?

stephanemagnenat, 2016-12-05:
Thank you very much for the PR, it is a great way to go forward!

> For erase and pop it must be a parameter.

If pop returns the popped elements, then the tuple size will be available as parameter. For erase, we need to pass it, and indeed it must be multiplied by the user as it as no idea of the size we are working on.

Here are the options I see for index multiplication:
- Nothing is multiplied. I do not really like it because it leads to writing somewhat complicated user code, although that if only `push` and `pop` are used, it is not a problem.
- One compromise we could have is to have `get`, `set`, `insert`, `erase`, `size` working without multiplication, but `push` and `pop` working with multiplication. Basically this reflects two level of abstractions in the API. If we are short of space, we could even not give access to `get`, `set`, `insert` and `erase` and only keep `push`, `pop` and `size`. However `size` still would be un-multiplied.
- Have everything multiplied excepted `erase`, which cannot infer the size. It is not very clean because it breaks convention in the middle of the API.

What do you think?

davidjsherman, 2016-12-05:
I think option two is the best API. I think in Push-Pop mode `size != 0` is the idiom for testing whether the deque is empty, and it doesn't matter that a nonzero number is the number of elements rather than the number of tuples.

I think we can fit it in 2.5 KiB: the loops in insert and erase can be replaced by two memcpy, and there are some duff-device like tricks if we can use to factor the code.

stephanemagnenat, 2016-12-05:
> I think option two is the best API

Good, I agree also. Should we keep `get`, `set`, `insert`, `erase` in the API? A compromise would be to keep them in a private API, like `deque._get`.

Also, do you agree that `pop_back/front` should copy the element before deleting it?

davidjsherman, 2016-12-05:
I would like to keep `get`, `set`, `insert`, `erase`, so if you think they are confusing then let's make them private.

> Also, do you agree that pop_back/front should copy the element before deleting it?

Yes, probably the users will expect that and as you say that gives us the size of the tuple.
(I hope Bertrand Meyer isn't following this thread. Mixing a transformer with an inspector!)

I don't want to optimize before the API is settled and the implementation is correct, but there are already a couple of things that could reduce the code footprint: with some stack munging (push and rotate) `push` and `pop` are just jumps to `insert` and `erase`; shifting one half of the deque right or left should be common static functions that call `memcpy` (twice because of modular arithmetic); `get` and `set` are also `memcpy` twice.

stephanemagnenat, 2016-12-05:
> if you think they are confusing then let's make them private.

I do not have a strong opinion, I was enumerating the options. Maybe to decide we need to think more: what are the use cases for them? What type of users will need them and when?

> Bertrand Meyer

I guess his systems have more memory than our poor little Thymio

> I don't want to optimize before the API is settled and the implementation is correct, but there are already a couple of things that could reduce the code footprint:

I agree, once we are clear with the API, we should write enough test cases to get convinced that the naive implementation is correct, then we can optimise.

davidjsherman, 2016-12-06:
In fact no multiplication is necessary in the Push-Pop API because it doesn't use numerical indexes.

All of the planned operations are in the latest commit to the PR (957af2bab128ce8f8fa98c10d7cd36d5b8f2ed70).

Nothing is yet done for failure tests and exception handling.

davidjsherman, 2016-12-07:
The Deque operations are ready for review. Commit b51fd7419c51ec2a8ca80b8a5dbde1daece556ad implements exceptions and includes tests of failure scenarios. See [tests/vm/data](https://github.com/davidjsherman/aseba/tree/583-vm-natives-deque/tests/vm/data) for the complete set of tests.

It looks like these changes would add 2430 bytes to the firmware, documentation strings included (comparison made using `xc16-gcc -mcpu=24FJ128GB106 -O1` and `xc16-ld -r -M`).

davidjsherman, 2017-01-25:
Implementation merged in 429af76935f45bcc0da57ddca6dffe413ff95cff

closed 2017-01-25 by davidjsherman