aseba-community/aseba #590: submodule removal problematic

aleb, 2016-12-09:
The instructions in https://www.thymio.org/en:sourceinstall say to run `git clone --recursive git://github.com/aseba-community/aseba.git` but that fails with
```
fatal: No url found for submodule path 'tests/externals/Catch' in .gitmodules
```
Because of this `clients/studio/plugins/ThymioBlockly/blockly` is empty and the build fails with: 
```
make[2]: *** No rule to make target '/home/aleb/dev/thymio/aseba/aseba/clients/studio/plugins/ThymioBlockly/blockly/tests/asebastudio.html', needed by 'clients/studio/qrc_asebastudio.cxx'.  Stop.
```

Maybe you should separate the initialization of the submodules from the git clone command.

stephanemagnenat, 2016-12-09:
I updated the documentation. On the long run, I plan to move the compilation instructions inside the repositories themselves so that they are linked to the correct branch.

aleb, 2016-12-09:
Good plan, we have the same for Pitivi.
I think the problem is still not fixed:
```
$ git clone --recursive git://github.com/aseba-community/aseba.git
Cloning into 'aseba'...
remote: Counting objects: 16939, done.
remote: Compressing objects: 100% (84/84), done.
remote: Total 16939 (delta 26), reused 0 (delta 0), pack-reused 16855
Receiving objects: 100% (16939/16939), 14.67 MiB | 8.07 MiB/s, done.
Resolving deltas: 100% (10562/10562), done.
Submodule 'clients/studio/plugins/ThymioBlockly/blockly' (https://github.com/aseba-community/blockly.git) registered for path 'clients/studio/plugins/ThymioBlockly/blockly'
fatal: No url found for submodule path 'tests/externals/Catch' in .gitmodules

$ cd aseba
$ git submodule update --init                                     
fatal: No url found for submodule path 'tests/externals/Catch' in .gitmodules

$ ls -l clients/studio/plugins/ThymioBlockly/blockly
total 0

```

stephanemagnenat, 2016-12-10:
I removed the --recursive for git clone in the documentation.

aleb, 2016-12-10:
It's not enough. "git submodule update --init" fails on a new clone. See
above.

stephanemagnenat, 2016-12-10:
You are correct and this behaviour surprises me. In the master branch, the Catch submodule is neither mentioned in .gitmodules nor in .git/config, so I do not know where git gets this path.

aleb, 2016-12-11:
Try `rm -rf tests/externals/Catch` and commit the removal?

stephanemagnenat, 2017-01-03:
> Try rm -rf tests/externals/Catch and commit the removal?

I think @davidjsherman  did that, but the problem is still here.

davidjsherman, 2017-01-03:
The Catch submodule was removed in 6d415e162ae8c35aef6036dfd8a87ce42ef99fa0.

The Jenkinsfile in fb37f0018c3d6d8ce0ff78b604148326e50f289b does a clean build and does use a `git submodule update --init`. It relies on a development environment defined by the Ansible scripts in [aseba-jenkins buildfarm](https://github.com/davidjsherman/aseba-jenkins/tree/master/resources/buildfarm).

So a thing to check would be whether the procedure in https://www.thymio.org/en:sourceinstall follows the same steps as the [Jenkinsfile](https://github.com/davidjsherman/aseba/blob/fb37f0018c3d6d8ce0ff78b604148326e50f289b/Jenkinsfile), and check that the prerequisites are equivalent to the environment built in [aseba-jenkins buildfarm](https://github.com/davidjsherman/aseba-jenkins/tree/master/resources/buildfarm).

aleb, 2017-01-03:
Maybe that commit is incomplete: https://github.com/aseba-community/aseba/tree/master/tests/externals
I suggested to remove the folder because after "rm -r tests/externals/Catch" I see "git status" shows the change as commitable.

stephanemagnenat, 2017-01-03:
Your are correct, it solves the problem, fixed in ed9acd2770b82676f38223308f4ad61dedab9b8d.

closed 2017-01-03 by stephanemagnenat