aseba-community/aseba #592: Move official CI to Inria

stephanemagnenat, 2016-12-15:
As agreed with David Sherman, Inria's CI will be used for Aseba in the future. To complete the move, we must do these actions:

* Replace ad hoc packager scripts with cpack,
* Write a master Jenkinsfile to automate build, test, package pipelines (see PR #581, aseba-community/dashel#20, and enki-community/enki#26),
* Configure Jenkins to also automatically compile pull requests.

Work is under way, this issue aims at tracking overall progress.

davidjsherman, 2017-05-09:
@aseba who should have administrative access to the build farm?

davidjsherman, 2017-05-10:
@aseba-community/mobsya who should have administrative access to the build farm?

mbonani, 2017-05-10:
in Mobsya it should be me now, it could evolve latter, Stephane is already I think
Thank

mbonani, 2017-07-04:
could we add Yves Piguet to ave access to the CI?
what is the procedure?

davidjsherman, 2017-07-04:
(PM reply)

mbonani, 2017-07-10:
I just exploring more the CI and was wondering what is the difference between "Aseba developer comunity" and "Github 3 Aseba" projects

davidjsherman, 2017-07-11:
"Aseba developer community" scans the aseba-community GitHub repository, looking for projects to build. Currently only **aseba** and **dashel** meet the criterion of having a Jenkinsfile in at least one branch. We could do the same for Enki. The "DJS" project scans my repository into order to test my own branches.

The three GitHub 1, 2, 3 projects were manually created.

I was waiting until the upgrade to Jenkins 2.60 to choose between the two, because I understand that there are changes to the GitHub Organisation plugin and I wanted to evaluate them first.