aseba-community/aseba #595: Compiler: more optimisation

stephanemagnenat, 2016-12-19:
It would be nice for the Aseba compiler to provide the common optimisations done in other compilers, such as common sub-expression elimination, etc. Some of these would be more effective if the language would provide local variables, as discussed in issue #161.

motib, 2016-12-19:
Compiler optimization is a difficult task and I don't see a need for it in Aseba programs. In my experience, loops (that normally benefit from optimization) often involve sampling multiple sensors and these loops wouldn't benefit from optimization.

In addition to local variables, what I would really, really like to see is the possibility of calling a subroutine during initialization. I find myself duplicating code for initialization and for re-initialization by touching a button. (There is probably an issue on this; if not, I can add one.)

stephanemagnenat, 2016-12-19:
> In addition to local variables, what I would really, really like to see is the possibility of calling a subroutine during initialization.

Commit c3ff4262ec56ab691903b8eae699baf9e7bf7e18 added this possibility. Version 1.5 and later do support it.

stephanemagnenat, 2016-12-19:
> Compiler optimization is a difficult task and I don't see a need for it in Aseba programs.

The problem is that VPL tablet, whose advanced mode supports visual state machines, generates complex AESL that easily fills up the memory. So some optimisations might be useful, such as deleting dead code more efficiently, implementing common sub-expression elimination, or removing useless assignment.

I totally agree that it is a difficult task, but at the same time it is so fun and rewarding, much better than meetings ;-)

stephanemagnenat, 2016-12-19:
> Version 1.5 and later do support it.

I updated the online manual in French and English.

motib, 2016-12-20:
I noticed that the manual doesn't discuss initialization at all, so I added a section on the topic. (English only, I don't trust the quality of my French...)

stephanemagnenat, 2016-12-20:
Thank you! I will update the French version as soon as I have some time.

stephanemagnenat, 2016-12-20:
Translation done (see https://www.thymio.org/fr:asebalanguage#toc17), thank you very much for your contribution and care to details :)