aseba-community/aseba #598: Native function `math.addscalar` could be removed by extending `math.add`

stephanemagnenat, 2017-01-03:
The function `math.addscalar` is used to add a scalar to a vector. But if `math.add(A,B,C)` could be extended such that if len(C) <= len(A) and len(C) % len(A) == 0, then C would be extended to len(A) by duplication and then the value added, then `math.addscalar` could be removed. The same logic could be extend for parameter B and also to other math functions having an array as destination.

FrancescoMondada, 2017-01-03:
Removing a function would make old code not compatible anymore...

stephanemagnenat, 2017-01-04:
> Removing a function would make old code not compatible anymore...

That is true, and I have no idea of how many people are using it. If only we had analytic on that...

But sometimes one has to clean up things, all programming languages do that, although they try to minimize the amount of change.

FrancescoMondada, 2017-01-04:
Cleaning bad things is useful, but I do not see the need to clean this up. Perhaps, at code level, it could be interesting to implement the chage you describe but keep both function (addscalar will redirect to add)

stephanemagnenat, 2017-01-04:
> I do not see the need to clean this up

The motivation to clean this up is that now that we had lengthy discussions of the behaviour of the deque library, I think that it would make sense from a usability and education point of view to generalise the existing vector functions, making addscalar obsolete. The latter function is a break of symmetry, which is something ugly we do not want to teach. Designing an API is an art that requires long thinking, so it cannot be right the first place.