aseba-community/aseba #600: Can not connect to Aseba DBus services!

Lunik, 2017-01-16:
Hi,
I'm trying to use the [python library for Thymio](https://github.com/aseba-community/aseba/tree/master/examples/clients/python-dbus) that you provide.

I have run `sudo asebamedulla "ser:name=Thymio-II"` to enable asebamedulla. There I can send code with aseba desktop interface.

But I'm trying to run python code: `sudo dbus-launch python aseba.py`. It should run the your default main but I get this error: 
```
Traceback (most recent call last):
  File "aseba.py", line 234, in <module>
    with Aseba(options.system) as aseba:
  File "aseba.py", line 70, in __init__
    raise AsebaException("Can not connect to Aseba DBus services! "
__main__.AsebaException: 'Can not connect to Aseba DBus services! Is asebamedulla running?'
```

stephanemagnenat, 2017-01-16:
I do not see why you launch the Python code with sudo. Most likely asebamedulla is not available from the sudoed user session.

Lunik, 2017-01-16:
That was my second try. However it doesn't work without anyway...

stephanemagnenat, 2017-01-16:
I suggest to try with an external dbus viewer app, such as `qdbusviewer` to see if you can see medulla. Which system/version are you using?

Lunik, 2017-01-16:
I'm using jessie/raspian 8.0

Lunik, 2017-01-19:
I finally find out where my problem was. 
asebamedulla and my python script where not on the same dbus deamon.
Starting them with the same work pretty fine !

closed 2017-01-19 by Lunik