aseba-community/aseba #605: Chinese version generate an error

FrancescoMondada, 2017-02-02:
Aseba Studio version 1.5.5 (build ver. git-b858c2e/protocol ver. 5) on MACOS 10.8.5, when choosing chinese as main language, generates an error message : "The help filter for the langauge "ja" has not been found. Falling back to the default language (en). This is probably a bug, please report it.". Then, studio starts in English but freezes (cannot get the descriptor of the robot that is connected ...?)

stephanemagnenat, 2017-02-02:
It works here with master version on Linux, I will test on OS X as soon as my work Mac is unpacked (we are moving).

stephanemagnenat, 2017-03-21:
With master branch, on OS 10.12.3, I have no error, and the help content is in English, as expected. I think this bug is corrected, but I'll keep it open until 1.6 is released, so that you can test by then and close it if the problem disappears.

mbonani, 2018-04-17:
seems to be fixed

closed 2018-04-17 by mbonani