aseba-community/aseba #608: Thymio not reconnecting with mac 10.12.3

sharon13m, 2017-02-23:
Hi,
I've tried with a couple different Thymio  and it is not reconnecting.  I have a 2013 mac running 10.12.3, a very new Thymio with serial starting with G, and have also tested with E, running firmware 10.  Aseba version 1.5.4 I was running, it didn't reconnect, so I just upgraded to 1.5.5 and it still does not reconnect.  Any help would be appreciated.
Thanks,
Sharon

stephanemagnenat, 2017-02-27:
I have testing with an iMac 27'' with the same version and re-connection works, with Aseba master branch (but there were no change related to connection since 1.5.5).

Could you please give some details about the situation when this re-connection problem appears? Are you using a wired or a wireless Thymio?

sharon13m, 2017-02-27:
Hi Stéphane, it is a wired thymio.  Nothing unusual that I can note that I'm doing.  How can I give you more information?  I'm happy to help assist as needed, there was nothing notable that I'm doing though, it just doesn't reconnect.

stephanemagnenat, 2017-04-10:
Do other wired Thymio work with reconnection? If so, maybe this one has an electric problem?

closed 2018-04-17 by mbonani