aseba-community/aseba #609: Support unicode names for variables, constants and events

stephanemagnenat, 2017-03-13:
The Aseba compiler is fully unicode compatible, allowing unicode characters in variable, constant and event names.

While the switch expects UTF8-encoded string values, it checks these using a dumb regex ("^[A-Za-z_][A-Za-z0-9_\\.]*"), which disallows unicode characters. It does so using C++'s own regexp support. A solution to this problem is to use instead the extended regexp grammar and the `[:alnum:]` entry (see http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap09.html#tag_09_04). However, for this to work the switch should force the use of UTF8 locale, which is unlikely to work on Windows. A fall back would be to convert the string to wstring before running the regexp. It is probably the safest solution.

ypiguet-epfl, 2017-06-19:
I couldn't find it. Where is it? Why not using the same code as in the compiler, if the test is already implemented there?

stephanemagnenat, 2017-06-19:
The new switch is in the https://github.com/aseba-community/aseba/tree/http-refactor branch, in /switches/http2, intermixed with old code.

The compiler is using a hand-made parser instead of regular expressions.