aseba-community/aseba #612: Errors with asebahttp and Thymio firmware 10

Okimi-, 2017-03-21:
Hello !
On windows, i'm using the asebahttp.exe command to send .aesl files to Thymio with this : 
`asebahttp.exe --aesl file.aesl "ser:name=Thymio-II"`

With a Thymio and the firmware 9.0, i haven't any problems, the file is transfered and run on the Thymio.
With a Thymio and the firmware 1.0, an error occurs and return this message
`1 scripts have no corresponding nodes in the current network and have not been loaded.`

Is there a correction to make to the aesl file to work with asebahttp and Thymio firmware 10 ?

mbonani, 2017-03-29:
Hello,
I am think you have to take asebahttp that is compile in 1.5.5 package. The aesl file should also be made with the targeted Thymio. Then there is perhaps an issue with the ID of the Thymio if you use a Wireless one. We manage changes in studio, but probably asebahttp is not. So if you have made the aesl file with one robot, it will not work with a second Wireless Thymio
regards

stephanemagnenat, 2017-04-10:
@Okimi-, any update using Aseba 1.5.5?

Okimi-, 2017-04-10:
No ! There isn't any update with Aseba 1.5.5...

*Okimi - Créateur de Blockly4Thymio*
*Blockly4Thymio, environnement de programmation pour le robot Thymio II.*

mbonani, 2017-05-08:
I tested and get same result of non working in the 1.5.5 that it is on our package and also on last compile in master branch.
It is working on David package you can find here: https://github.com/davidjsherman/inirobot-scratch-thymioII/releases/download/0.7.4-alpha/Aseba-1.5.3.Scratch-0.7.4b-win7.exe
Should ask David what the difference...

mbonani, 2018-04-17:
fixed

closed 2018-04-17 by mbonani