aseba-community/aseba #614: Integrating another language

Grahack, 2017-03-31:
Hi,

I'm the author of http://microalg.info and I would like to integrate it to Aseba.
Could you please give me hints or a kind of a roadmap for this task?
Or maybe this info already exists somewhere?

Thanks.

stephanemagnenat, 2017-03-31:
The main question is what type of integration you are thinking about? Do you want to generate AESL code? Do you want to do a Studio plugin? Do you want to do a standalone application?

Which language/framework microalg is using?

Grahack, 2017-03-31:
MicroAlg is embedded in http://picolisp.com which has different implementations: C, Java, JS and a custom ASM.
Generating AESL would be quite easy (MicroAlg already "exports" to several different languages), I'd like to do a Studio plugin.

stephanemagnenat, 2017-04-10:
Doing a plugin with Studio is relatively easy, if you know C++. For instance, VPL and Blockly do so.