aseba-community/aseba #615: Update codebase to C++11

stephanemagnenat, 2017-04-04:
The current Aseba codebase is mostly C++03. Recently, Aseba has been switched to C++11, and new code is expected to use the new C++11 features. Progressively, existing code should be migrated.

The following features should be used:

- **override**: methods that override parent ones should explicitly state that through the `override` keyword.
- **noexcept**: functions that are guaranteed not to return exception should state that through the `noexcept` keyword.
- **constexpr**: objects and functions that can be const expressions should state that through the `constexpr` keyword.
- **autogenm**: in interfaces with virtual destructors, when not manually specified, the following *auto-generated members* should be specified as `default`: copy constructor, copy assignment operator, move constructor, move assignment operator.

stephanemagnenat, 2017-04-05:
Commit 05d3b36710b0f09557925c0573873fecf8ed4aec updates `common/msg` to C++11 following these rules.

THS-on, 2017-04-17:
Commit 05d3b36710b0f09557925c0573873fecf8ed4aec breaks build process because in tests/CMakeLists.txt the directory "msg" is defined, but is missing.  
If the line is removed building works fine.

stephanemagnenat, 2017-04-18:
Commit c4c5307 adds the missing files, thank you for reporting the problem!

stephanemagnenat, 2017-09-22:
A large part of this can be done automatically, with supervision, using `clang-tidy`. With `FILENAME` as the cpp file to process, one can preview the changes with:
```clang-tidy -checks='modernize-*' -header-filter='.*' FILENAME -- -std=c++11```
and apply them with:
```clang-tidy -checks='modernize-*' -header-filter='.*' -fix FILENAME -- -std=c++11```

Additional include options should be given after `-std=c++11` when relevant.

stephanemagnenat, 2017-09-22:
Commit 9802ba91 updates `compiler`.

stephanemagnenat, 2017-09-22:
Commit 8b6e0f4e updates `common/utils`.

stephanemagnenat, 2017-09-22:
Commit 80f6bae7 finishes updating `common/msg`.

stephanemagnenat, 2017-09-22:
Commit daaf1be3 updates `common/zeroconf`.

cor3ntin, 2018-02-01:
Or you could use a compilation database

davidjsherman, 2018-02-01:
Is there a way to produce a compilation database that is compatible with Sonarqube? 
Currently we produce one using `build-wrapper` from [SonarCFamily for C/C++](https://docs.sonarqube.org/pages/viewpage.action?pageId=7996665) but it is sometimes a bit flaky and also closed-source.

cor3ntin, 2018-02-01:
@davidjsherman cmake does it, yes https://cmake.org/cmake/help/v3.5/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html

davidjsherman, 2018-02-01:
@cor3ntin: Do you know whether they are compatible? If you can generate one, I can test it with sonar-scanner.

How do you recommend managing the three different compilation databases for the three different build environments?