aseba-community/aseba #618: Size of blocks and fonts in Blockly

motib, 2017-04-10:
With computer screens having ever-higher resolutions, it is important that the blocks and fonts in the blocks either automatically adapt to the resolution or allow the user to choose the size. On a laptop with 2560x1440, the blocks were close to unreadable.

stephanemagnenat, 2017-04-10:
A hack on this could be to zoom the webview. But maybe when we switch to Qt5, this bug disappears, because it has proper HiDPI support so the web view should render the content already scaled up. Note that upgrade to Qt5 is planned for version 1.7.

davidjsherman, 2017-04-10:
Shouldn't need a hack, Blockly can scale smoothly.
Clicking on the (+) button in Thymio Blockly increases the size of the blocks.

stephanemagnenat, 2017-04-10:
That is true, we could probably initialize the zoom of current Blockly from the estimated DPI. Yet I would prefer invest energy porting Aseba to Qt5, which should give reasonable defaults.

motib, 2017-04-10:
Thanks, David. I didn't notice the +/- buttons on the lower right. But instead of closing the issue, I would like to ask that those icons not be displayed so dimly :-).

cor3ntin, 2017-12-07:
Blockly is being removed from studio.

closed 2017-12-07 by cor3ntin