aseba-community/aseba #619: Crash if an Array with 40 elements is sent.

tjenni, 2017-04-27:
If I execute the following programm, Aseba Studio crashes if N=40 or bigger. For N=5 the program works correctly.

# initialize map
var map[N]
var i = 0

while  i < N do
	map[i] = 1
	i = i + 1
end

# send map as an event
onevent button.left
emit sendMap map

stephanemagnenat, 2017-05-02:
Which target do you use? A Thymio, a simulated Thymio, a dummy node?

tjenni, 2017-05-02:
I use a Thymio II robot and Aseba runs on macOS Sierra.

stephanemagnenat, 2017-05-02:
This bug appears to be a Thymio wireless bug. I can reproduce it with Thymio Wireless using Linux and OS X, but not with the same Thymio using the USB cable or with dummynode.

It seems that messages of 40 elements or more are truncated into invalid Aseba messages.

stephanemagnenat, 2017-05-02:
As this is a Thymio wireless bug, a corresponding issue has been opened aseba-community/aseba-target-thymio2#22.

mbonani, 2018-04-17:
fixed

closed 2018-04-17 by mbonani