aseba-community/aseba #624: Robot in playground: provide default behaviour

stephanemagnenat, 2017-05-05:
Playground should provide a way to specify default robot behaviors through an AESL source code. Practically, there are different ways of doing so:
1. Add the source code of a given node as part of the text content of the robot tag.
2. Add a `sourceCode` attribute that links to an AESL file.

Solution 2. is more work as it requires a full AESL parser in Playground, but it might be the occasion of factoring the different versions these parsers throughout the Aseba codebase.

stephanemagnenat, 2018-01-15:
As a temporary solution, this is currently done by have `asebamassloader` loading an aesl file.

mbonani, 2018-01-31:
is the massloader solution sufficient?

stephanemagnenat, 2018-01-31:
Using massloader is a ugly hack in my opinion, so I would like to keep this issue open, but maybe it does not need to be solved for 1.7. However, I suggest to wait for deciding that once we have a clear idea of what should go in 1.7.

davidjsherman, 2018-02-01:
The second suggestion is cleaner but seems inconsistent when there is more than one robot. An Aesl file defines a *network*, so by associating an Aesl file with each node, we implicitly define independent networks. It seems to me that a Playground with more than one node should almost always run a Switch, so that nodes that can physically interact can also communicate.

Furthermore, an Aesl file is already a way to associate programs with nodes, so providing a second mechanism in a Playground file is redundant. The responsibility for sending the program to the nodes could be the responsibility of the Switch. The current Http switch already has this ability and we can resolve this issue using the current command-line options.

A further improvement would be to explicitly declare networks in the Playground file. I think this would improve clarity. It can be implemented now by spawning an `asebahttp` process for each network, but could in the future be implemented by adding Switch to the Playground (to avoid separate processes). For example a Playground with two networks could look something like this:

```
<aseba-playground>
  <world w="300" h="300" />

  <robot type="thymio2" x="100" y="100" nodeId="1" name="Thymio 0" />
  <robot type="thymio2" x="150" y="150" nodeId="2" name="Thymio 1" />
  <robot type="thymio2" x="200" y="200" nodeId="3" name="Thymio 2" />

  <networks>
    <network id="1" name="Network 1" aeslFile="thymio-default.aesl" />
    <network id="2" name="Network 2">
      <event size="2" name="Clock"/>
      <constant value="0" name="False"/>
      <node nodeId="2" name="thymio-II">...</node>
      <node nodeId="3" name="thymio-II">...</node>
    </network>
  </networks>
</aseba-playground>
```

The first `<network>` loads a network from an Aesl file, the second `<network>` is exactly the XML contents of an Aesl file.

An optimization, when there is only one node in the network, would be to imitate `asebamassloader` and send the program to the node without spawning a Switch.

davidjsherman, 2018-02-05:
PR #819 illustrates how to build networks using processes, and adds a test for the format extension proposed above. (The test is marked as expected to fail, because the extension isn't implemented.)

stephanemagnenat, 2018-02-05:
Your point is interesting. I think that we see emerging two use cases/views of what is an Aseba network:
1. Originally, an Aseba network was within a single robot (marXbot), to connect multiple micro-controllers.
2. Later, especially with Thymio, came the idea of putting multiple Thymios in the same network using identifier remapping.

From this realisation, there are several things to consider:
- The `nodeId` cannot be a property of a robot, as a robot might be composed of several nodes. Also, to simulate accurately a robot (for example marXbot), the node ids should match the real robot, as codes typically depend on them.
- I do not agree that a playground with more than one node should always run a Switch, because in the context of multi-player games (à la challenge), different people connect to different robots, as in reality.
- My suggestion of having the `sourceCode` attribute was to address 1. In a lot of basic cases this would be sufficient.
- I am open to the idea of `<networks>` but we should agree to the previous points first.

davidjsherman, 2018-02-05:
Thanks for the clarification, I didn't know all of that. I had always assumed that the node id remapping was a hack to get around Thymios always having the same node id. So if I understand you, the Aseba bus was never conceived as a way to have multiple independent robots communicate? Two marXbots on the same bus would be able to send a message directly to their peers' microcontrollers.

stephanemagnenat, 2018-02-05:
The node id remapping is a hack indeed. And yes, originally Aseba was never conceived to have multiple robots communicating, it was conceived to have parts of a modular robot communicating together, assuming each part would have a different node id and no parts will be present twice on the robot. Aseba was conceived in the context of a swarm robotics project, where inter-robot communication would typically be local and embodied, for example through vision, sound, local infrared communication, and physical interactions.