aseba-community/aseba #629: Add support for pseudo-terminal in asebaswitch

ypiguet-epfl, 2017-05-15:
Improvement suggestion: add an option to asebaswitch to accept a single connection on a pseudo-terminal.

For a clean implementation, this would require a new protocol in Dashel, e.g. serin, which creates a master/slave pair of pseudo-terminals, and a method to get the slave port ("serin:" would create a slave pty, e.g. "/dev/pts/35", and the slave Dashel target would be "ser:device=/dev/pts/35"; similarly for "tcpin:port=40000", the counterpart would be something like "tcp:port=40000").

Code for pseudo-terminals in Linux:

#define _XOPEN_SOURCE	// for pseudoterminal in stdlib.h
#include <stdlib.h>
#include <unistd.h>
masterfd = open("/dev/ptmx", O_RDWR | O_NOCTTY);
if (masterfd < 0)
	error...
if (grantpt(masterfd) < 0)
	error...
if (unlockpt(masterfd) < 0)
	error...
printf("ser:device=%s\n", ptsname(masterfd));	// slave
// communication over masterfd...
close(masterfd);

stephanemagnenat, 2017-05-15:
For which use case is this needed? On Posix, Dashel's `SerialStream` class inherits from `FileDescriptorStream`, adding support for fetching by serial port by high-level name and line settings. The `FileStream` class inherits also from `FileDescriptorStream`, without doing this additional machinery. Therefore, can we use the `file:` protocol, for instance `file:/dev/pts/35`?

davidjsherman, 2017-05-16:
If the tty doesn't work, it should work to create a fifo file in Unix, e.g.
```
mkfifo spy
asebaswitch file:spy
```

ypiguet-epfl, 2017-05-16:
Stéphane, /dev/pts/35 is created on demand, as a side effect of opening /dev/ptmx. The goal would be to test client applications which use the Aseba protocol over what looks like a serial port (ser:xxx in Dashel).

David, pseudo-terminals work in both directions. I don't think a fifo would work between Aseba Studio and Thymio, for instance: a process doesn't want to read back what it has written itself.

stephanemagnenat, 2017-05-16:
@ypiguet-epfl I see the use case more clearly now. I am not sure that this is generic enough (clearly not working on Windows for instance) to be included in Dashel mainline, but feel free to create a branch with it, for test purposes and to further discuss this option.

ypiguet-epfl, 2017-05-17:
I've forked Dashel and Aseba at https://github.com/ypiguet-epfl/aseba
To test it in Linux, connect a Thymio, switch it on, check its device path with `ls /dev/ttyACM*`, start asebaswitch with something like `./switches/switch/asebaswitch --pty -d -l -v 'ser:device=/dev/ttyACM0'`; it will display the corresponding slave device as a Dashel name (e.g. `ser:device=/dev/pts/43`). Launch Aseba Studio and specify that name in Custom.