aseba-community/aseba #631: RC codes in advanced VPL missing

FrancescoMondada, 2017-05-17:
Some IR remote controls (for instance microspot 4 in 1), in the middle of the arrow pad, have an OK button with code 13. This should be added to the existing code 87.

stephanemagnenat, 2017-05-17:
This make sense. Do you envision other code/cases to add? Maybe we should ask on the mailing list? I would like to collect all changes in one commit. I think that we can deliver that for version 1.6.

FrancescoMondada, 2017-05-17:
It’s difficult to survey all remote controls, the community could help, yes...

> 
> This make sense. Do you envision other code/cases to add? Maybe we should ask on the mailing list? I would like to collect all changes in one commit. I think that we can deliver that for version 1.6.

------------------------------
Prof. Francesco Mondada
Mobile robots architecture & design

Ecole Polytechnique Fédérale de Lausanne (EPFL)
Laboratoire de Systèmes Robotiques (LSRO)

Postal address:
Francesco Mondada 
EPFL  - STI  - IMT - LSRO 
bureau MEB3426, Station 9
CH-1015 Lausanne, Switzerland

tel: +41 21 693 73 57
fax: +41 21 693 78 07
http://lsro.epfl.ch

stephanemagnenat, 2017-05-18:
I've notified the mailing list. The decision of whether to add a new code should result from a balance between supporting more devices, and avoiding the generated code to become too large.

stephanemagnenat, 2017-09-26:
As the mailing list is down, let us collect here the new codes we want to add.

mbonani, 2017-09-26:
I think it should react like Thymio firmware does:

```
switch(vmVariables.rc5_command) {
			case 2: //front arrow
			case 80:
			case 32: 
				if(rc5_speed_t)
					rc5_speed_t = 0;
				else
					rc5_speed_l += RC5_SPEED_STEP;
				break;
			case 4://left arrow
			case 85:
			case 17:
			case 77:
				rc5_speed_t = -RC5_SPEED_STEP;
				break;
			case 8: //back arrow
			case 81:
			case 33:
				if(rc5_speed_t)
					rc5_speed_t = 0;
				else
					rc5_speed_l -= RC5_SPEED_STEP;
				break;
			case 6: //right arrow
			case 86:
			case 16:
			case 78:
				rc5_speed_t = RC5_SPEED_STEP;
				break;
			case 5: //center button
			case 87:
			case 13:
				rc5_speed_t = 0;
				rc5_speed_l = 0;
				break;
		}
		vmVariables.rc5_command = 0;
```

here it also include the normal number, in VPL it should then in my sense focus arrow pad.

stephanemagnenat, 2017-09-26:
I agree about having the same behaviour of the firmare.

stephanemagnenat, 2017-09-26:
Fixed by cbd23954.

commit cbd23954 (2017-09-26): Updated code for RC buttons, fixes #631
closed 2017-09-26 by stephanemagnenat