aseba-community/aseba #635: Why class Zeroconf does not inherits from class std::vector<Target>?

stephanemagnenat, 2017-05-23:
Class `Zeroconf` has a member `targets` of type `std::vector<Target>`, and recopies a significant part of the interface of `std::vector<Target>`. Is this encapsulation useful? Why not inheriting publicly from `std::vector<Target>`?

stephanemagnenat, 2017-05-23:
A related question is why, in class `Zeroconf`, the wrappers for the members of `std::vector<Target>` are virtual?

marvelous, 2017-05-23:
Is Zeroconf a vector? Sorry to bikeshed here, but wouldn't it be cleaner to expose the `targets` member with a reference getter and remove all these methods?

stephanemagnenat, 2017-05-23:
> expose the targets member with a reference getter

Or just make it public. @davidjsherman what was the rationale behind the current design?

stephanemagnenat, 2017-06-09:
Currently changing the code to make it public.

stephanemagnenat, 2017-06-23:
Changed in commit 5fa41768fcbead37f5ce01f49c39c5d976d148ad.

closed 2017-06-23 by stephanemagnenat