aseba-community/aseba #639: Bytecode corrupted when resetting "when" flags

ypiguet-epfl, 2017-05-29:
In https://github.com/aseba-community/aseba/blob/master/vm/vm.c at line 726, AsebaVMSendExecutionStateChanged should reset the "when" flag only in ASEBA_BYTECODE_CONDITIONAL_BRANCH instructions, not in the parameters of multi-word instructions or in the initial event dispatch table.
Test case: in Aseba Studio, run
var foo = -24064
The variable is set to -24576 (0xa000) instead of -24064 (0xa200) because 0xa200 is wrongly considered to be an ASEBA_BYTECODE_CONDITIONAL_BRANCH and its "when" flag, 0x0200, is cleared.
Short-term solution: in AsebaVMSendExecutionStateChanged, iterate in the bytecode starting after the event dispatch table and increment pc by the actual instruction size.
Long-term solution: store "when" flags in data, not in code, to keep code constant.

stephanemagnenat, 2017-05-29:
Oups, it's a hard one. Thank you very much for pointing it out. I agree with the short-term solution, but I would implement it in `AsebaVMResetWhenFlags`. For the long term, if I had to redesign Aseba VM now, I would do it completely differently (even more lightweight VM with all functions as native) which would make this problem disappear as the flag would naturally be stored in data.

stephanemagnenat, 2017-05-29:
Commit b1ad652 adds a unit test for this problem.

closed 2017-05-29 by stephanemagnenat