aseba-community/aseba #642: Serving local files from asebahttp and asebahttp2

ypiguet-epfl, 2017-06-19:
Suggestion: add support to asebahttp and asebahttp2 to also serve local files, to avoid cross-origin issues. A basic implementation with a single directory and a few mime types (html, js, css, png, jpg) would already be very useful.

stephanemagnenat, 2017-06-19:
Agreed, it makes sense.

davidjsherman, 2017-06-19:
I'm not sure I understand. Could you describe a test that would demonstrate this? Some URIs and their expected results would help.

There is already a plan to serve the OAS apidoc and the thymio-motion.aesl file, I am guessing that this a more general service?

ypiguet-epfl, 2017-06-22:
David, I'm thinking about web applications. Browsers usually restrict cross-origin requests with `XMLHttpRequest` (as I understand it, currently asebahttp2 is completely open with `Access-Control-Allow-Origin: *`). Another advantage is to avoid separate http servers or reverse proxys.

I'm not aware of development plans. Is there any link you can share?

davidjsherman, 2017-06-22:
I don't have my notes on the single-page app at the office, but can dig them up. They would be PDFs extracted from my livescribe. I am sure that they weren't made public in GitHub. Plans for a single-page app drove the definition of the new asebahttp API.

Historically the CORS modification in asebahttp was made in f45c83c038e03fe99a98a8b46e76eaeed2493bf1 in response to a discussion with Pierre Rouanet, but there was no issue opened for it. About that time there was a discussion about embedding the `thymio_motion.aesl` file in a firmware version of asebahttp. In part to avoid cross-origin but also to permit the Thymio to act as a wireless AP.