aseba-community/aseba #656: Upload artifacts to a web site using credentials binding

davidjsherman, 2017-07-03:
Currently PR #581 for Jenkins automation is on hold, waiting for the final step on the checklist: uploading the compiled artifacts to the distribution site. I suggest that we move that requirement to a separate issue.

There are three different situations:
1. Official Aseba releases. A version tagged **release-*** could be uploaded to https://github.com/aseba-community/aseba/releases using a command-line tool like [github-release](https://github.com/aktau/github-release).
1. Nightly builds and PRs. These could be uploaded to the existing archive of nightly builds. (For those developers who don't already have access to the CI platform.)
1. Official Thymio.org releases. These should be put together by the Thymio team and may include components that aren't a part of [aseba-community](https://github.com/aseba-community/aseba). So these artifacts should not be uploaded by the Jenkins automation.

davidjsherman, 2017-07-03:
Note that, presently the Jenkinsfile in #581 only performs extended tests and packaging for **master** branches. We would want **release** branches to be packaged, too.

Does anybody use the packaged nightly builds? It seems to me that a build for each branch (including PR branches) would be more useful for testing. Maybe conduct a survey on the *aseba-dev* list?

stephanemagnenat, 2017-07-03:
> Does anybody use the packaged nightly builds?

You mean from master branch? If so, yes, everyone testing the simulator for instance.

> Maybe conduct a survey on the aseba-dev list?

It does not exist any more as gna is down forever. Hence, we will have to discuss on github or create a new mailing list (we can create an issue for that if we think it is useful to have such a list).