aseba-community/aseba #660: Thymio VPL (standalone) text code erase by selecting and typing enter

cbarraud, 2017-07-07:
In Thymio VPL Standalone, when you create a program and have the text displayed on the side, if you select with your mouse part of text and click on the "Enter" key, the selected text is erased. It doesn't do that with any other key I tested. 

If you do this program for example:
[image]

and then select some text like the green light call, when you type "Enter", it erases. 

[image]

If you click then on "Play" in VPL, the program downloaded is the one without the green light. 

If you change one single thing in the program (adjust a color, change a button...) the text is retyped. 

I don't know how it is managed, but it could be a good idea to forbid the use of the "Enter" key here, or you can mess up with your code :)

Thymio VPL Version 1.5.98
[image]

Computer with Windows 8.1

stephanemagnenat, 2017-07-07:
Fixed by commit b7135ce.

closed 2017-07-07 by stephanemagnenat

cbarraud, 2017-07-07:
That was quick 👍