aseba-community/aseba #662: Updating ROS integration to protocol 5

wasowski, 2017-07-07:
First, thank you for all your work on Aseba and Thymio.  I'm considering using Thymio for teaching at IT University of Copenhagen, and explore options of connecting to it.  Right now, I am trying to get the Thymio ROS driver to work (http://wiki.ros.org/thymio_navigation_driver) -- also authored by you. Thanks!

I noticed that in late January you have removed the DescriptionsManager class in a commit that implements the version 5 of the protocol (which is necessary for the newest Thymio's).  Unfortunately, this breaks the Thymio ROS driver, which used this class.  I am looking into updating the driver, but obviously, without expertise in Aseba (and not so great in ROS...) I am wandering around blindly.  Could you hint how the driver should be updated? Is there somewhere a similar code that you have updated that one could use as  a  guiding example? Any help will be hugely appreciated!

PS. I am asking in this repo, as your ROS repo is inactive (and in anyway, I am working of a later fork that catkinized the build).

wasowski, 2017-07-08:
For the record this is my work in progress: https://github.com/wasowski/ros-aseba/commit/8748a8f153acc6636dbf5f6bc3ff52a39cd56201 . It compiles with Aseba 1.5.5 but since other things in the project still don't work, I cannot really tests if this is sufficiently correct.

stephanemagnenat, 2017-07-10:
Dear @wasowski , indeed `DescriptionsManager` was renamed `NodesManager`, as now it is also responsible for taking care of connecting and disconnecting nodes. An example of how to use it can be found in `examples/clients/cpp-shell`. This should be the only change to do. To use `NodesManager`, subclass it and implement at least `sendMessage()` so that it can request node descriptions and ping nodes for activities. Function `pingNetwork()` should be called regularly, typically every second. Every message received should go through `processMessage()` so that the node manager can be aware of what is happening on the network. You can subclass the other protected virtual classes if you are interested in them, but you do not have to.

Indeed ros-aseba is inactive as I have not been using ROS for years, but I'll happily merge changes that update it if it is useful for the community.

Thank you very much for your contributions!

wasowski, 2017-07-11:
Is `pingNetwork` a new function? I can't see it being called anywhere in the old driver.

stephanemagnenat, 2017-07-11:
Yes it is new, because the new protocol supports the disconnection of nodes at any time (for instance if one stops a wireless Thymio), there is a need to periodically ping the network to see who is still here, and who joined recently.