aseba-community/aseba #665: blockly do not launch correctly in master branch on windows

mbonani, 2017-07-10:
[image]

stephanemagnenat, 2017-07-10:
Interesting, could it be that the Qt you use to compile does not have WebKit support enabled?

mbonani, 2017-07-10:
It is from the EPFL buildfarm. Also the same on my computer, both have the msys2 compiler installed following the compile.Windows.md instructions

mbonani, 2017-07-12:
After Stephane investigation the problem come from the qtwebkit compiled from the new compiler. We have differents solutions:
- work to make the webbase solution (actual sandbox) more usable
- recompile QT 4 with the new compiler
- pass to QT5
We will make a small survey to get data from users.

cor3ntin, 2017-12-07:
Blockly removed from studio

closed 2017-12-07 by cor3ntin