aseba-community/aseba #670: multiple tabs with Thymio wireless

FrancescoMondada, 2017-07-16:
On MACOS, the aesl program in attachment is generating a new tab at each disconnection of the same wireless thymio, creating a multitude of tabs, as illustrated by the screen copy.

[bug-multi-wireless.zip](https://github.com/aseba-community/aseba/files/1151310/bug-multi-wireless.zip)
[image]

stephanemagnenat, 2017-07-16:
This aesl has multiple times a node with the same name and the same nodeId, which makes it an invalid aesl program, although this is not detected by Studio. How was this program created?

FrancescoMondada, 2017-07-16:
From what I know, the students who created this code changed once their Thymio wireless and copied the program from a tab to another, but this should not create this problem. At one point their tabs started multiplicating, no idea in which exact conditions... sorry.

stephanemagnenat, 2017-10-11:
I managed to reproduce this bug here, I will investigate.

stephanemagnenat, 2017-10-12:
Actually the Linux problem was due to a change in Qt5. Do you experience the reported problem with wired Thymio on reconnection?

riedo, 2017-12-01:
I experienced this problem today with a wired Thymio, Windows 10, latest aseba dev version. 
While testing reconnection, the robot suddenly connected in a new tab, the first one staying blank. This happened on the second reconnection. 
Unable to reproduce unfortunately, now Studio always crashes on the second reconnection.

mbonani, 2017-12-06:
probably the problem are different in macosx and windows. I notice a unstability on reconnection on windows machine, also when you have a wireless and the robot is not there for a moment. We have to test after the pull request that correct the dashel bug is merge

stephanemagnenat, 2017-12-06:
Knowing that @FrancescoMondada is on macOS, I think that the specific bug of this issue is present on macOS and Linux. We should try to find a way to reproduce it in these systems.

cor3ntin, 2017-12-06:
@mbonani I spent a fair amount of time trying to reproduce it last night on linux,  both with valid and invalid aesl files, without success. Can you see if you can come up with reliable repro ?

mbonani, 2017-12-07:
on linux it happens only on test on qt5. In macosx I will try to reproduce it this afternoon when I am at EPFL

stephanemagnenat, 2017-12-07:
> on linux it happens only on test on qt5. 

Did you try the latest version of qt5 branch?