aseba-community/aseba #672: Crash of Asebahttp

cbarraud, 2017-07-21:
Hello

I was testing ScratchX with Elisa and Joel when my asebahttp crashed completely. I took a printscreen here:

[image]

It is not easy to reproduce though... I was just manipulating the ScratchX program when it crashed. I will try to find a way to reproduce that.

I was using

- Wireless Thymio, firmware 10
- Windows 8.1
- Chrome
- aseba-bin-1.5.98-git-84f1a55-win32

davidjsherman, 2017-07-21:
I think “trying to wait on more than 64 events” is aseba-community/dashel#14: there is a bug in the Windows implementation that will overflow the internal buffer. See [my comment](https://github.com/aseba-community/dashel/issues/14#issuecomment-205939981).

More generally, I agree that asebahttp needs to be made more robust. We lost precious time yesterday tracking down a connection error.

cor3ntin, 2018-04-17:
That was fixed in 1.6

closed 2018-04-17 by cor3ntin