aseba-community/aseba #673: Why is there no delay function?

fmatter, 2017-08-06:
I figure it would be pretty straightforward to implement a 'delay' or 'sleep' command, no? I don't see why the 'timer' automatically restarts itself. This way you can't just tell the robot to do something for e.g. 4 seconds, you have to stop the 'timer' again.

davidjsherman, 2017-08-06:
Do you mean a blocking wait, where the robot ignores events during the sleep period?

fmatter, 2017-08-27:
No, not really. I just mean something like

`call leds.bottom.left(0,0,32)
sleep(4)
call leds.bottom.right(0,32,0)`

, so it would first turn on the left led, then it would wait 4 seconds, and then it turns on the right led. With the timing capabilities there are right now, one would have to first set the "timer" to an interval of 4 seconds, and then turn it off again once the right led is activated to achieve this result.

davidjsherman, 2017-08-27:
The behavior you are describing is in fact not easy at all, because you want the robot to remember the instruction right after the `sleep`, so that it can continue execution once the four seconds have passed, while still responding to other events in the meantime.

An operating system with a preemptive scheduler can do this, of course, because it just puts the one process (or thread) to sleep while the others are executing. But the Aseba VM doesn't have a preemptive scheduler. It actually doesn't even have interrupts: events act as if they are interrupts because their handlers are always very short (10 ms is an eternity in PIC24 time) and always terminate before the next event triggers.

For the sake of discussion, let's try to hack this together without preemptive scheduling. Your `sleep` instruction would have to
1. record the address of the next instruction for the timer handler, and
1. exit from the code block so that other events can be handled.
1. Furthermore, when the timer rings, jump to the recorded address.

**But**, conceivably you could have more than one `sleep` in different places in the program. There is no way to forbid that, syntactically. So you have to record all of the addresses. Now, when the timer rings, you have to jump to each of them in turn, to wake up all of the threads. To make sure that no pending wake ups are missed, whenever a code block exits, you need to check the list of addresses to see if anything is waiting, so that you can wake it up before handling any new events. So we would still have to change the scheduler. ☹️ 

If `sleep` is globally blocking, rather than just blocking the thread, then we don't have to remember any addresses because there is only one place in the code to continue when the timer rings. But with this semantics, the robot is totally unresponsive during `sleep`.  ☹️

fmatter, 2017-09-21:
> But the Aseba VM doesn't have a preemptive scheduler.

Well that complicates things, of course. I see that it would not be as straightforward as I thought.

Still, a globally blocking `sleep` function might still be appreciated by users.

davidjsherman, 2017-09-21:
Thanks for your input.

I'm closing this for now, but will re-open if some use cases can be defined. A good use case would ideally consist of an exercise statement for students, and a teaching goal whose solution requires a blocking `sleep`.

closed 2017-09-21 by davidjsherman

stephanemagnenat, 2017-09-21:
An implementation of the transformation of a sleep statement over the Aseba language is available as an example in [`examples/switch/botspeak`](https://github.com/aseba-community/aseba/tree/master/examples/switches/botspeak).