aseba-community/aseba #675: Native function math.muldiv may produce rounding errors

stephanemagnenat, 2017-08-07:
Native function `math.muldiv` may produce rounding errors, for instance 2/3 is 0 rather than 1. It might be interesting for most end-user applications for `math.muldiv` to actually round correctly the numbers. To resolve this problem, one should:

- [ ] Write unit tests exhibiting the problem
- [ ] Implement rounding (https://github.com/aseba-community/aseba/blob/master/vm/natives.c#L821), by testing the sign and pre-adding half of the divisor to the dividend.

marvelous, 2017-08-07:
2/3 is 0 for some rounding modes. My guess is that the rounding mode should be explicitly documented.

stephanemagnenat, 2017-08-07:
That is true, it should :)

The reason we were considering changing it, is that the current rounding introduces a bias (there are more values rounding to 0 than rounding to other numbers), which we think might introduce errors on operations accumulating a lot of results of muldiv with values close to 0, such as odometry.

retornaz, 2017-08-07:
This is a wrong excuse. You just need to handle the math operation correctly in odometry computation. The way it is implement is efficient on a microcontroller. if you redefine it to have rounding to the closest you will have perf. issues.

stephanemagnenat, 2017-08-07:
Well, the whole question is what is the scope of use of `math.muldiv`. The Aseba VM is slow compared to native code and calling a native function entails a significant overhead. Changing the rounding would probably not lead to a dramatic drop in performances, especially on small arrays. But I agree that it all depends on which use one wants to do of them.

mbonani, 2017-08-07:
We are working on a 16bit integer microcontroller and aseba unique variable are integer. In this case the result is only the quotient. Making the rounding different than most other language is for me not logical. 
Yes it should be documented.

retornaz, 2017-08-08:
rouding different than most other languages ? it's the C based rounding, that most of the programming language are using (at least for positive number).

stephanemagnenat, 2017-08-08:
Agreed, changing the title into a documentation issue.