aseba-community/aseba #681: correctly show contributors

mbonani, 2017-08-17:
As the project becomes bigger with several programs and tools (different clients, plugins) it becomes difficult to correctly make an authors pages. It is also perhaps to general and its is difficult to show in what the contributors really worked on. 
I open this issue to get your opinion on that. 
Should we multiply authors.txt or authors.md in the several client, plugins, switches to correctly keep trace?
Should we have a definite header in each file with rules to keep them update (a merge is only accept if header is update)?
Should we have something else ?

davidjsherman, 2017-08-18:
A related question is whether you want contributors to explicitly sign a CLA. Then you would have legal documentation as well as a database of contributors. Some of Inria's projects for example use the [CLA Assistant](https://github.com/cla-assistant/cla-assistant#contributor-license-agreement-assistant) integrated with GitHub's PR workflow.

mbonani, 2017-09-06:
Probably CLA is good for legal issue. I am more interest in the database that it can provide and how to show it. How to correctly keep trace of contributions and how to thanks them.
I will more invest this tool.

stephanemagnenat, 2017-09-06:
Indeed the two issues, CLA and authorship recognition, are different.

I see one value of CLA, to clarify who is the project maintainer, however it is not trivial, as currently I am the maintainer as a person and Aseba has no organization behind it. It is not clear whether assigning the licensing rights or the copyrights to me as a person makes sense. We could also make a new association for Aseba, but I do not see enough value compared to the added administration. An option could be to use the new company Hannes and me are in process of creating.

On the down side of a CLA, it might make contributions harder and more administrative, hindering the grassroot open-source approach I try to push for Aseba. Even the KDE project uses CLA optionally, to ensure a maximum of contributions. So I am note sure it is a critical element for Aseba now, as the LPGL v.3 is a very balanced and reasonable license.

Now for per-tool author contributions, it is an interesting thing to consider it indeed. What I think could be nice is to have a main authors.md file, compiled into a class that can report contributions, and for each tool to simply list the emails of the persons who contributed to this tool and its dependencies.

Maybe we could automatize the collection of contributions from the git log. It raises some issues, as all automatic metrics, such as the number of lines commited, or the number of lines of a given author in a given version, do not represent the work with fidelity. Maybe such a tool could be an aid to make sure we do not forget contributions.

stephanemagnenat, 2017-12-05:
I am now working on generic dialogue box as well as an underlying way to represent authors and contributors, and display them. That should factorise all code related to that in a single place.

stephanemagnenat, 2017-12-06:
New implementation of about box in Studio in branch [new-about-dialogs](https://github.com/aseba-community/aseba/tree/new-about-dialogs).

stephanemagnenat, 2017-12-07:
Commit d34f6d0fec0f89cbfc027b4afd48cbdb276d0116 adds support for parsing authors.txt and displaying the authors in the dialog box, filtered by the current application (actually supports multiple modules).

stephanemagnenat, 2017-12-07:
Commit 63a2acdd68195a0178386a8f64a6f974135f3487 implements the agreed solution and a new dialogue box for Studio.

stephanemagnenat, 2017-12-07:
I think that command-line tools are not critical for 1.6, so I am creating a new issue #731 for them and closing this one.

closed 2017-12-07 by stephanemagnenat