aseba-community/aseba #682: MacBook Pro 2016 Adapter wireless connection issue

tiemes, 2017-08-31:
A friend of mine cannot use the wireless dongle, as it seems to establish a connection and immediately breaks again, in a cycle. connection/lost/connection/lost in short few-second intervals. Both with Aseba Studio and VPL. 
He's using a LMP USB-C mini Dock, as well as a LMP USB-C multiport Adapter VGA & USB.
No problems with tethered USB cable connection. Tried with different Thymio II wireless.

MacBook Pro, 13Zoll, 2016, 
macOS Sierra Version 10.12.6

mbonani, 2017-08-31:
I will try to test but my first impression is that the highspeed USB-C is disturbing the RF part of the Wireless dongle.

stephanemagnenat, 2017-09-04:
Agreed, it is very unlikely that this problem is related to Aseba.

closed 2017-09-04 by stephanemagnenat

tiemes, 2017-09-05:
So will the Wifi Dongle not be usable with any USB-C device?

mbonani, 2017-09-05:
It works for me with this adaptor: Dell DA200 with an Samsung A5 (2017).
Something that could help is to have an USB extension cable that put the wireless module not to close of the adaptor.  I will make other test when I  get USB-C device. I have no Mac now to test.