aseba-community/aseba #692: Updating a TXT field by advertising an existing target does not work

stephanemagnenat, 2017-09-28:
Commit 0012396f adds support for the busy flag in dummynode, that should be set to true (and thus present in the TXT flag) if the target is being used.

While the call `Zeroconf::advertise()` is called with the updated `TxtRecord`, the description received by the client (for instance `asebatargetlist-thread`) is not updated.

stephanemagnenat, 2017-09-28:
This can easily be seen by launching a dummy and observing it, for example on macOS with:
```
dns-sd -B _aseba._tcp .
dns-sd -L "Dummy Node 0" _aseba._tcp .
```

davidjsherman, 2017-09-28:
The unit tests in [branch zeroconf-unit-tests](https://github.com/davidjsherman/aseba/tree/zeroconf-unit-tests) need to be finished (and updated wrt the new Zeroconf interface?).

stephanemagnenat, 2017-09-28:
Fixed by commit cc0c5112.

closed 2017-09-28 by stephanemagnenat