aseba-community/aseba #694: Feedback on GUI connection

cbarraud, 2017-09-28:
I find this new connection interface pretty neat! :))

I am just afraid it might be a bit technical for teachers in terms of which is which. Differentiate the local simulated robots from the USB or Wireless ones is easy when technical data such as "local serial port" doesn't scare you
￼
[image]

My only criticism would be that we could have a more user friendly GUI like with buttons or anything.

I made a quick drawing, more explicit than words. It's ugly, but you will get the jizz.

[image]

What do you think? This could become a wish.

stephanemagnenat, 2017-09-28:
The idea of using categories is interesting, but complications too much the generic Aseba GUI, as there might be a lot of different categories of robots.

If the combination of this generic GUI and the single connection to the first Thymio is not enough, I suggest to make a Thymio-specific dialogue within the Thymio launcher. That would also filter out things such as Bluetooth ports, etc.

cbarraud, 2017-09-28:
> I suggest to make a Thymio-specific dialogue within the Thymio launcher

I like the idea. It might also be more clever to do that in a more specific Thymio part such as the Thymio launcher instead of the generic Aseba launcher.

stephanemagnenat, 2017-09-28:
I've changed the title to reflect our discussion.

mbonani, 2017-09-28:
Just to come back at the original discussion two generic categories could be "local" (local robot connected), and "on network", and put alos robot of local simulator in first category

stephanemagnenat, 2017-09-28:
It is indeed a good idea, to try to resolve the hostname and see if it is on the machine.

stephanemagnenat, 2017-12-05:
I'm closing this in favour of #728, a launcher-specific issue.

closed 2017-12-05 by stephanemagnenat