aseba-community/aseba #697: Thymio Web Bridge blocked by Antivirus

paolorossetti, 2017-10-02:
A professor from Turin wrote me this request for support for aseba-bin-1.5.98-git-c010fd9-win32.exe that is recognized as a virus from Windows.

“aseba-bin-1.5.98-git-c010fd9-win32.exe  è tenuto prigioniero nel cestino dei virus e non vedrà mai più la luce. Esegui una scansione per essere certo che non abbia causato danni.”
Su altro computer con antivirus Symantec il programma è stato scaricato, ma alla richiesta di esecuzione l’antivirus l’ha cancellato dalla cartella dei downloads.

Both Avast and Symantec forbid to execute the exe file. 

How to solve this? any clue?

mbonani, 2017-10-02:
Hello,
For Avast we see it, I just send to them that is a false positive.  We will see if it resolved the problem. 
when we release we will check that the installer is ok for all virus. A good tool to test is https://www.virustotal.com/#/home/upload

For development version it could that people have to disable their antivirus temporarily

stephanemagnenat, 2017-10-03:
I wonder whether this problem would be away if we could sign the Windows package.

mbonani, 2017-10-03:
I got contact with avast and get this answer:

> Thank you for your reply.
> 
> I'd suggest checking these articles about Avast virus policy:
> - Avast Clean Guidelines: https://www.avast.com/faq.php?article=AVKB228
> - Avast File Whitelisting: https://www.avast.com/en-us/faq.php?article=AVKB229
> 
> It is always beneficial to use a valid digital signature on executable files. You can also submit your files for whitelisting before every new released in order to prevent detections.

mbonani, 2017-10-03:
signature is already in pipeline https://github.com/aseba-community/aseba-windows/issues/3
I add to find the better solution to make it. 

I also forgot: normally now "asebahttp" is not recognized as virus by avast. So Symatec has probably the same feature to provide whitelisting.

mbonani, 2017-12-14:
signature on windows packet are done, this problem should not appear anymore on latest build.
@paolorossetti do you think your contact can test it?

mbonani, 2018-04-17:
packet are signed should work

closed 2018-04-17 by mbonani