aseba-community/aseba #698: ASEBA studio crashes on dialog box

FrancescoMondada, 2017-10-02:
Studio in version 459 on MACOSX 10.12.6 : started aseba playground with R2T2 environment, started studio and after a small modemt, crash:

Process:               asebastudio [46594]
Path:                  /Applications/aseba-dev2/*/Aseba Studio.app/Contents/MacOS/asebastudio
Identifier:            org.mobsya.asebastudio
Version:               1.2
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           asebastudio [46594]
User ID:               501

Date/Time:             2017-10-02 14:21:39.527 +0200
OS Version:            Mac OS X 10.12.6 (16G29)
Report Version:        12
Anonymous UUID:        3769195E-9C4F-0275-4CF5-4F35B1473C92

Sleep/Wake UUID:       15D1B89A-5B95-4D4F-B6D5-065F06B140BE

Time Awake Since Boot: 240000 seconds
Time Since Wake:       7700 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type std::out_of_range: map::at:  key not found

stephanemagnenat, 2017-10-02:
Are you using the development version? It might be a duplicate of #696.

FrancescoMondada, 2017-11-09:
#696 is closed but this bug is still present.

stephanemagnenat, 2017-11-09:
> #696 is closed but this bug is still present.

No-one created a build since my commits that close #696. The last build is from October 24 (see http://mobots.epfl.ch/data/aseba-build-server/osx/master/)

mbonani, 2017-11-09:
the build is now working. I just launch it manually and it works...

FrancescoMondada, 2017-11-09:
Why are new builds triggered manually? Why one cannot test the last build downloading it from thymio.org?

FrancescoMondada, 2017-11-09:
I confirm, not crashing anymore.

closed 2017-11-09 by FrancescoMondada