aseba-community/aseba #699: See wrong Thymio when asebahttp is launched

FrancescoMondada, 2017-10-02:
With release 459 of Studio: 
- I start asebahttp and connect it to the USB port ("./asebahttp 'ser:device=/dev/cu.usbmodem211' -s 33333" 
- I start ASEBA Studio: the port 33333 does not appear, the "cu.usbmodem211" appears but cannot connect, no error display and target windows just reappear and ask to choose again.

stephanemagnenat, 2017-10-02:
zeroconf support for asebahttp is work in progress, see #690.

FrancescoMondada, 2017-10-02:
OK, but it is possible to avoid displaying connection that cannot be connected because used by another program? Or display an error?

stephanemagnenat, 2017-10-02:
For serial port it is not possible, as there is no way of knowing whether the port is being used without connecting to it, and we do not want Aseba connecting and sending data to random serial ports just to see if by chance they are Aseba targets.

A Thymio-specific dialogue box from a Thymio launcher could safely ignore everything that does not explicitly look like a Thymio.

closed 2017-10-03 by stephanemagnenat