aseba-community/aseba #703: Compilation on Debian Jessie x86 : nor VPL, nor launchers

ljoets, 2017-10-08:
Hello,
On a computer with Debian Stretch x64, I installed "aseba_1.5.5-stretch-amd64.tar" found here : https://www.thymio.org/fr:linuxinstall and everything is ok.
On my son computer, I compiled aseba on Debian Jessie x86 following this : https://github.com/aseba-community/aseba/blob/master/compile.Linux.md
Aseba seems to be installed, I haven't seen any errors, I can manually launch asebastudio, but I have no launchers (I use Gnome), and I can't find VPL :-(

stephanemagnenat, 2017-10-08:
Could you show me the configuration log? (the output of the cmake command for Aseba).

ljoets, 2017-10-08:
`ljoets@eeeBox-Malo:~/aseba/build-aseba$ cmake ../aseba -DCMAKE_BUILD_TYPE=RelWithDebInfo -Ddashel_DIR=../build-dashel -Denki_DIR=../build-enki
-- Git executable found
-- Git repository found
-- libudev found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Found ZeroConf in /usr/include;/usr/lib/x86_64-linux-gnu/libdns_sd.so
-- Enabling Dashel socketcan plugin
-- SDL2 found, enabling joystick/joypad support
-- Could NOT find PROTOBUF (missing:  PROTOBUF_LIBRARY PROTOBUF_INCLUDE_DIR) 
-- Google Protobuf library and compiler *not* found, usage logging disabled.
-- Qwt found
-- qcollectiongenerator executable found
CMake Warning at tests/compiler/CMakeLists.txt:186 (message):
  zzuf executable not found! Disabling fuzzy tests

CMake Warning at tests/compiler/CMakeLists.txt:196 (message):
  valgrind executable not found! Disabling memory leaks tests

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ljoets/aseba/build-aseba`

ljoets, 2017-10-08:
Ok, so I juste learn to read...
`sudo apt-get install zzuf valgrind`
...

ljoets, 2017-10-08:
So I can now launch `thymiovpl` in a terminal, and it works.
But I have no launcher in Gnome. Do I create it manually ?

stephanemagnenat, 2017-10-08:
Did you do a make install? Normally it should install the freedesktop entries. There might be a problem under Gnome (I only tested with KDE), but as these should be standard I would be surprised.

ljoets, 2017-10-08:
Yes, I dis a make install.

stephanemagnenat, 2017-10-11:
Indeed this is a missing feature in the current install script. The freedesktop entries are not installed by make install, but by the packager script (deb or rpm).

stephanemagnenat, 2017-10-11:
Commit a10bfc5 corrects the missing installation step. Could you please try again with the latest master branch?

ljoets, 2017-10-11:
Yeah ! Launchers and icons are here !
Many thanks :-)

stephanemagnenat, 2017-10-11:
you are welcome! Thank you for reporting the problem :)

closed 2017-10-11 by stephanemagnenat