aseba-community/aseba #705: Provide an AESL file containing the default behaviour of Thymio

stephanemagnenat, 2017-10-13:


stephanemagnenat, 2017-11-08:
@davidjsherman had the idea that if massloader can launch a file once, then the existing mechanism of playground to launch third-party command could be used to load a file by default. It is not ideal as it would show some garbage in the log of playground, but it could be a temporary fix.

stephanemagnenat, 2017-11-08:
Commit 3d104508 adds support for loading a code only once with asebamassloader.

mbonani, 2017-11-22:
[all-behavior-for-simulator.zip](https://github.com/aseba-community/aseba/files/1495381/all-behavior-for-simulator.zip)

Here is all behavior that are usable in the simulator. I let the attentive mode only in the menu but it do nothing.  Where should I add it in aseba? Do we add it only in packaging as a dependency?

stephanemagnenat, 2017-11-22:
I will fix some minor things (indentation, space consistency) and add it the the playground examples. Then we can try to let Playground load it through asebamassloader. We should make sure in the packaging that asebamassloader is available in the same directory as asebaplayground.

stephanemagnenat, 2017-12-05:
I cleaned the code up and added it, see commit 91c1ab9e.

@mbonani could you please further test? I feel that with the explorer behaviour sometimes the robot turns too fast and stays too far away from obstacle.

mbonani, 2017-12-06:
The motor target where not bounded, I correct this in my pool request.
But there is still a difference with the C code because in will not overflow like the aesl code:

```
	long temp1 = 0;
		long temp2 = 0;
		temp1 += vmVariables.prox[0];
		temp1 += vmVariables.prox[1] * 2;
		temp1 += vmVariables.prox[2] * 3;
		temp1 += vmVariables.prox[3] * 2;
		temp1 += vmVariables.prox[4];
		temp2 += vmVariables.prox[0] * -4;
		temp2 += vmVariables.prox[1] * -3;
		temp2 += vmVariables.prox[3] * 3;
		temp2 += vmVariables.prox[4] * 4;
		vmVariables.target[0] = speed - __builtin_divsd((temp1 + temp2) * speed, 2000);
		vmVariables.target[1] = speed - __builtin_divsd((temp1 - temp2) * speed, 2000);
		
		if(vmVariables.target[0] < -600) 
			vmVariables.target[0] = -600;
		if(vmVariables.target[1] < -600)
			vmVariables.target[1]= -600;
		if(vmVariables.target[0] > 600)
			vmVariables.target[0] = 600;
		if(vmVariables.target[1] > 600)
			vmVariables.target[1] = 600;
```

also I notice some strange value comming from the simulator like here
[image]

the real robot will never add some value 4 ,1, 3.
If it see nothing it should be zero

stephanemagnenat, 2017-12-06:
Aseba provides a native function called `math.muldiv(A, B, C, D)`: _Compute multiplication-division using internal 32-bit precision: Ai=(Bi⋅Ci)/Di. An exception will be triggered if a division by zero occurs._

This should allow you to avoid overflow. Btw, to which PR do you refer to?

stephanemagnenat, 2017-12-06:
> the real robot will never add some value 4 ,1, 3. If it see nothing it should be zero.

Ok, indeed the behaviour is different, because Enki still adds the noise to values out of range. I'm looking at a fix right now.

stephanemagnenat, 2017-12-06:
> the real robot will never add some value 4 ,1, 3. If it see nothing it should be zero.

Fixed in commit ea325a17.

davidjsherman, 2017-12-06:
In `thymio_motion` I did this with `math.dot` if I remember correctly

mbonani, 2017-12-19:
fix in https://github.com/aseba-community/aseba/commit/9999b0065510774196327c59b82198f8f0d52d37

mbonani, 2017-12-19:
how to make the default behavior charge at beginning, should we add something in the .playground?

stephanemagnenat, 2017-12-19:
Yes, provided that the `asebamassloader` binary is in the same directory as playground (or linked there), one can use the `process` tag to call `asebamassloader` to load this file. You can have a look in `targets/playground/example/unifr.playground` to see how to use the process command.

mbonani, 2017-12-19:
it does not work if I do in the file:

>  "process command=":asebamassloader --once ./examples/thymio-blockly-standalone.aesl tcp:localhost" 

not working

by hand you have this:: 

> C:\Program Files (x86)\AsebaStudio>asebamassloader.exe --once ./examples/thymio-
> default-behaviours.aesl "tcp:localhost;33333"
> ! 1392 bytecodes loaded to target thymio-II, you can disconnect target !

 but simulator do not accept charge it:

[image]

on real thymio it works

> C:\Program Files (x86)\AsebaStudio>asebamassloader.exe --once ./examples/thymio-default-behaviours.aesl "ser:name=Thymio"

stephanemagnenat, 2017-12-19:
Why did you use `./examples/` as path for the AESL?

stephanemagnenat, 2017-12-19:
I re-open as there is now a non-working AESL file.

stephanemagnenat, 2017-12-19:
Playground should change directory to the scene path.

stephanemagnenat, 2017-12-19:
Fully implemented by 565bdb17 and b6eb0117. All discovery kits and the empty Thymio scene have the default behaviour.

closed 2017-12-19 by stephanemagnenat