aseba-community/aseba #707: Error when building on Linux with parenthesis in path

ypiguet-epfl, 2017-10-19:
When building 20fee1c on Linux with
```
cmake .
make
```
when there is an opening parenthesis in the path (`/home/xxx/Downloads/aseba-master (2)`), make stops here:
```
[ 46%] Generating aseba-doc.qch
/bin/sh: 1: Syntax error: "(" unexpected
clients/studio/CMakeFiles/rendervplblocks.dir/build.make:219: recipe for target 'clients/studio/aseba-doc.qch' failed
```
clients/studio/CMakeFiles/rendervplblocks.dir/build.make:219 is
```
@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --blue --bold --progress-dir="/home/xxx/Downloads/aseba-master (2)/CMakeFiles" --progress-num=$(CMAKE_PROGRESS_4) "Generating aseba-doc.qch"
```
There is no error there if I rename `aseba-master (2)` as `aseba-master-2`.

stephanemagnenat, 2017-10-19:
It looks like the path with parenthesis is correctly enclosed by "", but somehow within CMake this creates a problem. Thus I am labelling this issue as "foreign issue". Ideally it should be reported to CMake itself.

cor3ntin, 2018-01-23:
Quoting won't help.
Have you tried:

`@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --blue --bold --progress-dir="/home/xxx/Downloads/aseba-master \(2\)/CMakeFiles" --progress-num=$(CMAKE_PROGRESS_4) "Generating aseba-doc.qch"`

?